FreeOpcUa/python-opcua #721: opcua.common.utils.NotEnoughData: Not enough data left in buffer

bryankim96, 2018-10-27:
I was working on a custom python OPC UA client when I started to run into issues when subscribing to many nodes at once. Although I know it's not recommended to create large numbers of subscriptions, for the sake of illustration I came up with this fairly minimal example. The code recursively traverses a tree of nodes, creates subscriptions, and monitors all property and variable nodes. 

When I run this code it succeeds in creating the first few subscriptions, but then runs into a NotEnoughData exception. I saw a number of other issues related to this NotEnoughData exception, but none of them seemed directly related. 

Do you have any ideas as to what might be causing this problem? Is it possibly a server-side issue?

Code snippet:

```python
from opcua import Client, ua

class SubHandler(object):

    def datachange_notification(self, node, val, data):
        print("OPC UA: New data change event", node, val)

    def event_notification(self, event):
        print("OPC UA: New event", event)

# Function to recursively traverse node tree
def __traverse_node(node, opcua_client, seen):

    if node in seen:
        return
    else:
        seen.add(node)

    subhandler = SubHandler()

    for n in node.get_variables() + node.get_properties():
        print("Subscription created.")
        sub = opcua_client.create_subscription(1000, subhandler)
        sub.subscribe_data_change(n)

    # Recurse through object children
    children = node.get_children(nodeclassmask=1)
    if children:
        for n in children:
                __traverse_node(n, opcua_client, seen)

if __name__ == '__main__':

    opcua_client = Client(SERVER_ADDR)
    opcua_client.connect()

    device_tree_root = opcua_client.get_objects_node().get_child(["2:DeviceTree"])

    seen = set()
    __traverse_node(device_tree_root, opcua_client, seen)

```

Trace:

```bash
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
Subscription created.
exception calling callback for <Future at 0x7f7e51af5da0 state=finished returned Buffer>
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/client/ua_client.py", line 402, in _create_subscription_callback
    response = struct_from_binary(ua.CreateSubscriptionResponse, data)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 502, in struct_from_binary
    val = from_binary(uatype, data)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 483, in from_binary
    return struct_from_binary(uatype, data)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 502, in struct_from_binary
    val = from_binary(uatype, data)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 479, in from_binary
    return unpack_uatype(vtype, data)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 197, in unpack_uatype
    return st.unpack(data)
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 138, in unpack
    return struct.unpack(self.format, data.read(self.size))[0]
  File "/home/bryankim96/.virtualenvs/psct_gui/lib/python3.6/site-packages/opcua/common/utils.py", line 65, in read
    raise NotEnoughData("Not enough data left in buffer, request for {0}, we have {1}".format(size, self))
opcua.common.utils.NotEnoughData: Not enough data left in buffer, request for 4, we have Buffer(size:0, data:b'')
```

Also, I'm on the most recent version of python-opcua, installed directly from source.

bryankim96, 2018-10-29:
> At least creating a high number of requests is a good way to find errors.... A possible method to find out if this is a client or server error is to monitor session with Wireshark and see if there is any corrupt data packets that Wireshark cannot parse

OK, let me look into this and try to send you some wireshark captures. 

> What server is this?

The server was written using the United Automation OPC UA C++ SDK, but the server-side code is a little long/complicated (and I don't think I can share it freely). Let me see if I can come up with a more minimal example that reproduces the error.

> And are you using special objects like extended node ids?

No, the server doesn't use any extended node ids. Are there any other common causes of parsing errors that I should look for?

zerox1212, 2018-10-29:
I think a few datatypes have more customer binary parsers. Maybe UUID and datetime.

oroulet, 2018-10-30:
@bryankim96 I cannot remeber one parsing error for several years (apart from a small thing in expended node ids), so I would be surprised if we had one, but on the other hand, Unified Automation is not known to have many bugs either... so this is a very interesting case to debug....

closed 2018-11-13 by bryankim96

Chris-Tophski, 2020-04-20:
Is this not on the agenda any more? I'm running into the same issue:

We have a Siemens HMI with its standard OPC server installed here and we are perfectly able to read, browse, analyze and change values, however, data change subscriptions don't work in FreeOpcUa. Also, there is nothing special, just a few hundred named standard nodes set up by TIA. As far as I can see, the subscription methods don't raise an exception at least, but when the value should be read, the NotEnoughData error occurs (i. e. Buffer.read() method line 65). There seems to be no node, for which it works.

I also tried UaExpert, with which I can connect and draw items from the Address Space panel to the Data Access View. The items are correctly read and updated, although I constantly get 'Call Republish failed (ret = BadUnexpectedError)' errors. Has anyone experience with this kind of behavior? Maybe it is related to the NotEnoughData error somehow?

WastlSepp, 2020-10-16:
@Chris-Tophski did you get a clue on what was wrong there?

I'm running into the same issue right now.
Having a Selogica HMI with OPC server installed here and not being able to subscribe to more than 8 nodes.

UaExpert working fine, but with FreeOpcUa i get the NotEnoughData error.