FreeOpcUa/python-opcua #723: xmlexporter raises exception on node that lacks optional attribute

mtx500, 2018-10-30:
common/xmlexporter.py uses functions from common/node.py to read attributes from nodes. These functions raise ua.StatusCodeError if not Good. But there is the code BadAttributeIdInvalid which is returned by the server if an attribute is not available. And this code is returned when an optional attribute is not present, so this is actually not an error.
I added locally in xmlexporter the statements try: and except BadAttributeIdInvalid: around these places, but in my opinion it would be better if the implementation in node.py would not raise an exception for this error code but instead return a "None" value instead.

oroulet, 2018-11-07:
I have not looked at code but adding try catch in exporter sounds. This is a place we export things without really knowing what they are and things may fail

oroulet, 2018-11-07:
send PR