FreeOpcUa/python-opcua #724: Client timeout doesn't trigger

pinkynrg, 2018-11-03:
I'm explicitly setting the Client timeout like so:

```
client = Client('some connection string', timeout=2)
client.connect()
```

But in some cases, when the connection string is not correct, the Client hangs forever.

How can I solve this?

xeden3, 2018-11-27:
make following change to fix it
    file://python-opcua/opcua/client/ua_client.py
    line:139
    sock = socket.create_connection((host, port))
    to
    sock = socket.create_connection((host, port), timeout=self.timeout)

    code..
    def connect_socket(self, host, port):
        """
        connect to server socket and start receiving thread
        """
        self.logger.info("opening connection")
        sock = socket.create_connection((host, port), timeout=self.timeout)
        # nodelay necessary to avoid packing in one frame, some servers do not like it
        sock.setsockopt(socket.IPPROTO_TCP, socket.TCP_NODELAY, 1)
        self._socket = ua.utils.SocketWrapper(sock)
        self.start()

Henna1977, 2018-12-12:
Solution works great.
I got the same problem with the timeout.

Can you create a pull request on your solution?