FreeOpcUa/python-opcua #727: Question : SourceTimestamp not working for external clients

Toppert93, 2018-11-08:
Whenever I update a certain value of a variable from an external client, the source Timestamp becomes 1/1/1601 (like the Server Timestamp, but this is less important...).

I tried with different clients (2 different from Node-Red and also with UAExpert). Maybe the timestamp I'm sending is wrong, but my feeling is telling me than it would not work like this and some kind of handler would need to be written to handle incoming values and write them in python as client to the server including its own timestamp?

I anted actually to replace an OPCUA Server on one side with this one, but it will be impossible if the timestamp is not adapting.

MichaelKorn, 2019-03-19:
+1 I have the same issue :-/

zerox1212, 2019-03-19:
I would take a look at https://github.com/FreeOpcUa/python-opcua/blob/a8553adb23a040ea64e171f6ae4c2d4d22356b98/opcua/ua/ua_binary.py#L36 . This library converts the incoming binary to python `Datetime` from what I remember. Whatever the client is sending must not be formatted how the code expects it.

MarcusWalther, 2019-06-07:
I have the same issue with UaExpert - what is the solution?

oroulet, 2019-06-07:
As far as I remember we donnot write this attributes. but if you write it explicitly for example in python:
dv = ua.DataValue(2)
dv.SourceTimestamp = datetime.now()
myvar.set_value(dv)

then you will see it updated from uaexpert or any other client

if you are using an external client and do not see that value, then it measns that you client do not specify that attribute

MarcusWalther, 2019-06-10:
Thank you @oroulet for the fast reply. I'm just wondering, because the timestamps were correctly set in opcua-0.98.1 (tested with python 3.6 and the latest version of UaExpert). But it is not working with the latest opcua release (neither with python 3.6 nor with 3.7). So I assume that there was a change in the time stamp handling of the library.

[image]

oroulet, 2019-06-10:
Yes before we updated automatically both timestamps, but it was a performance issues and we found out many servers were not doing it. So I removed that code

MarcusWalther, 2019-06-12:
The whole discussion seems to be related to #850. Maybe you can add the functionality again as "optional feature", so that we have the possibility to turn it on for the server.