FreeOpcUa/python-opcua #730: Server react on read node

IzonFreak, 2018-11-14:
I have a physical device that has many parameters. I want to display all of this parameters on an OPC UA server and the client should be able to read/write the parameter. For write I know you create a subscription but for a write I can't find a similar mechanism, I can't cycle read each of them all the time since that would be impractical and take too much time, I just want to read the parameter on command, so that I can get the value from the device and then return it to the client.

     Client---read---->Server---read--->Device

    Client <---resp---Server <---resp---Device

Could you please point me in the right direction ?

zerox1212, 2018-11-14:
There are several ways to do what you want probably.

The easiest thing I can suggest is to create UA methods on your device object in the server. `read_param` and `write_param`. When client calls this method on the device the server can read/write the device.

IzonFreak, 2018-11-15:
Hello Andrew, 

I already followed the mirror example ! I actually did exactly that with my server. I would prefer if I could find a way to react to reads on a variable node, methods might be a little bit unintuitive for a user but that is definitely an idea that I could propose. 

Best Regards,
Ricardo J.

IzonFreak, 2018-11-15:
Hello Andrew,
I already followed the mirror example ! I actually did exactly that with my server. I would prefer if I could find a way to react to reads on a variable node, methods might be a little bit unintuitive for a user but that is definitely an idea that I could propose.

zerox1212, 2018-11-15:
Intercepting a read request from the client might be difficult. @oroulet would have to help you on that one. 

Other people on here have essentially asked the same thing in the past and I find it to be a valid, although kind of strange design, for OPC UA. To do it you will likely have to really go deep into the library.

IzonFreak, 2018-11-15:
The problem comes if you are trying to tunnel old devices, that are slow. If you device uses some kind of serial communication for example, you will need to create the package send it to the device and then wait for the answer. 
You can cycle read the most use parameters, but if you want to share all the capabilities it would not be possible since the device will take to much time answering so many requests.

oroulet, 2018-11-15:
There is/was a value callback stuff. Reading a value would trigger a method call. The bag side is that it breaks subscription but maybe it is no problem for you? The value_callback stuff is still I address_space.py but code is needed to make usable from server object. Try to grep for value_callback in code