FreeOpcUa/python-opcua #731: SourceNode vs. emitting Node for events (e.g, Audit-Write-Events)

Nebukadneza, 2018-11-16:
Hi there,

when I wanted to send AuditWriteUpdateEvents for a Variable-Node in my AddressSpace I ran into some issues. Let’s first clarify what the spec wants us to do:
- Part 3, 4.6.1: Objects and Views can be used to subscribe to Events. The EventNotifier Attribute of those Nodes identifies if the Node allows subscribing to Events.
- Part 5, 6.4.2, BaseEventType: "SourceNode identifies the Node that the Event originated from [...] Some subtypes of this BaseEventType may define additional rules for SourceNode."
- Part 3, 9.5, AuditEventType: "For example, in response to a Client issuing a write to a Variable, the Server would generate an AuditEvent describing the Variable as the source [...]"
- Part 5, 6.4.24, AuditUpdateEventType: "The SourceNode for Events of this type shall be assigned to the NodeId that was changed."

To me, that means that only Objects and Views can emit events. The prime example is the Server-Object-Node. While BaseEventType abstractly describes the SourceNode as »where the event originated from«, the AuditEventType and AuditUpdateEventType are more explicit: For them, SourceNode shall be the node of which the value/attribute/whatever was changed. This especially means that they have to have a Value, thus be a Variable. However, Variables cannot have EventNotifier Attributes — thus, they cannot be the emitting node.

To me, this sounds like that often, the Node emitting the Event is not identical to the SourceNode referenced in the Event itself. Specifically, a Variable which got its value updated could be the SourceNode, and the Server-Object-Node could be the emitting node.

As far as i see in the EventGenerator and MonitoredItemService, this is not how it’s implemented currently. Currently, the EventGenerator guesses the SourceNode to be the emitting node — and fixes it up for the user. Subsequently, the MonitoredItemService will emit the Event from it’s SourceNode-field.

My proposal to redeem the situation — always given that my spec-interpretation is correct — would be to add a pure-python attribute (not a opcua-attribute/property) to the BaseEvent, which carries the emitting-node/nodeid. Based on this, the MonitoredItemService can send the event originating from the correct node, while still describing another node — possibly a Variable — in the SourceNode (opcua)-property.

Please let me know what you guys think about my ramblings ^_^.
Thanks and Best Regards
-Dario

oroulet, 2018-11-16:
I did not understand everything but as far as I remember, in theory the spec allows any object nodes to emit events, but in practice most servers and most clients only suport the Server node as emiting node. (we supporta any node with the correct attributes) So python-opcua set the Server node as default for events. Concerning the source node you can put whatever you want in there depending on the context.

zerox1212, 2018-11-16:
From what I remember only BaseEventType is currently implemented. To generate an AuditEvent the user would have to implement it on their own I think?

Are you suggesting to make a PR that implements AuditEvents, and thus this new functionality of defining source node? The reason I ask is that I think this library is doing BaseEventTypes correctly.

oroulet, 2018-11-17:
Yes I do not thing that AutditWriteUpdateEvent is implemented but you can probably implement it by looking at the other events and add whatever logic you want there

Nebukadneza, 2018-11-19:
Hi,

manually emitting an Audit*Event currently isn’t hard — you just fiddle the fields of your event together, hand it to the generator, and trigger away. As for automatic generation, e.g. on value-changes, I’m not sure in what way that could be implemented so users can take advantage of its benefits without being domineered over …. I guess that’s something to talk about in a next step. For the moment, people implementing Servers can easily generate and trigger such events by hand.

Regarding Source-Node vs. Emitting node: If you look at Part 5 6.4.2, the BaseEventType has a SourceNode field of type NodeId. This field can be set to any arbitrary NodeId — be it an Object, Variable, View, etc. This SourceNode, however, is not necessarily the Node from which the Event was emitted. Most often Events are emitted by the Server-Node — but even though the Server-Node emits an AuditWriteUpdateEvent, the SourceNode of this event would point to some variable that has changed — for example some SetPoint our server represents, or similar.

This last functionality is what’s not implemented correctly at the moment: Currently, the EventGenerator and downstream-functions assume that the SourceNode-field of the Event describes which Node *emits* the Event. However, SourceNode is not the emitting Node, but the Node that is referred to as the source — the emitting Node is different.

I think for clarity I‘ll whip up a PR to fix this soon, and also provide an example to generate AuditWriteUpdateEvent, so other users can already profit from the manual way of triggering these ….

Best & Thanks,
— Dario

oroulet, 2018-11-19:
@Nebukadneza OK. you are probably right. I kind of remember wondering about these two properties when writing that code. Great if you can make a PR