FreeOpcUa/python-opcua #734: Not support kepware?

tvvbbb, 2018-11-20:
I want to import node with kepware. When I import node, kepware show me that it can't browse the node.

oroulet, 2018-11-20:
Kepware has a very bad UA support but it should work.

tvvbbb, 2018-11-20:
In fact, it can't work. Any software can change UA server to DA server instead of Kepware?

oroulet, 2018-11-20:
There are many gateways. Justgooglr

tvvbbb, 2018-11-20:
I just find kepware. Can you introduce me some?

zerox1212, 2018-11-20:
ProSys, Matrikon, Software Toolbox Tunneler. There are many.

tvvbbb, 2018-11-20:
Thank you very much.

closed 2018-11-20 by tvvbbb

oroulet, 2018-11-20:
https://www.google.com/search?q=opc+ua+gateway. Prosys has very good us support for example. But you can also use python openopc to make your own gateway

tvvbbb, 2018-11-20:
Is easy to make gateway? I use Cogent DataHub to covert but it still not work. Cogent DataHub show error :
[2018-11-20 16:39:00.201] E: [UA] : UA Client: Attach Failed. Name='OPCUA001', Endpoint='opc.tcp://192.168.1.217:4841/freeopcua/server/': BadNotImplemented: The requested operation is not implemented.

tvvbbb, 2018-11-21:
I want to know how to make my own gateway.

zerox1212, 2018-11-21:
You glue this library and a DA library together to read/write from a single list of nodes.

smalhao, 2019-09-06:
Hello,
I would like to ask if there is an alternative to open source Kepware, dare not to pay for the license.
I'm sorry to ask here but I don't know in which section of Github I should ask.

tvvbbb, 2019-09-10:
Sorry, I don't know.

oroulet, 2019-09-10:
kepware does not do anything else than translating one protocol into another one so Python itself is an interesting alternative. just use the libraries you need for the protocol you needs

smalhao, 2019-09-10:
maybe I should explain what I want, it's the following, I have an OPC UA Server on my network and I have the Local Discovery Server service installed on a PC and the KEPWARE installed on the same PC so that through the KEPWARE address I can access the OPC UA Server through the list of servers in KEPWARE, the idea is for a user to access the list of OPC UA Servers through the KEPWARE and select the OPC UA Server that wants to see the sensor data that it has coupled to it.

My question is if there is some software that makes me discover the OPC UA Server and show the list of them, only without having to pay software license, because the KEPWARE has paid license.

zerox1212, 2019-09-10:
The CLI can do it. Look at the code for CLI command `uadiscover`. Then write a GUI to show the results.

smalhao, 2019-09-10:
zerox1212 can you explain, i´m not understand. The CLI of KEPWARE?

zerox1212, 2019-09-10:
Look at the README. The python-opuca library has a CLI and can find UA server via discovery. You can make your own service for listing UA servers. Other than that I do not understand what you are doing.