FreeOpcUa/python-opcua #767: No handlers could be found for logger "opcua.client.client"

Test4564, 2018-11-29:
Hello,

i have a Problem with the example from the OPC UA Server (Python 2.7).
If i start the script i get the text:

No handlers could be found for logger "opcua.client.client"

Have anyone a idea, how I can fix this Problem?

Thank you very much!

Test4564, 2018-11-29:
The code which I use:

import sys

sys.path.insert(0, "..")
import time

from opcua import ua, Server

if __name__ == "__main__":

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://192.168.178.49:4230/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    # get Objects node, this is where we should put our nodes
    objects = server.get_objects_node()

    # populating our address space
    myobj = objects.add_object(idx, "MyObject")
    myvar = myobj.add_variable(idx, "MyVariable", 6.7)
    myvar.set_writable()  # Set MyVariable to be writable by clients

    # starting!
    server.start()

    try:
        count = 0
        while True:
            time.sleep(1)
            count += 0.1
            myvar.set_value(count)
    finally:
        # close connection, remove subcsriptions, etc
        server.stop()

zerox1212, 2018-11-30:
I'm not 100% sure with the info you give, but make sure you python is up to date. If you are on a Raspberry Pi or something you could have an extremely old version.

Test4564, 2018-12-02:
Thank you!
@zerox1212 the Script is running on my laptop...
I use now Python 3.7... and it solved the Problem!

I used the version 2.7 because I like the old print "function"...

Test4564, 2018-12-02:
How can I close the topic?

oroulet, 2018-12-02:
Click close button?

closed 2018-12-02 by oroulet

jdenninghoff, 2022-11-14:
@oroulet Although this topic was closed, i am facing the same issue. The problem is, that i am forced to use a Python 2.7.X Version on the client side. On github (https://github.com/FreeOpcUa/python-opcua) it says that the Client and Server work on Python 2 and 3, is that not the case for Python 2 then? I have been testing it in an virtual environment with Python 2.7.18 
[image]