FreeOpcUa/python-opcua #770: Couldn't Write Value To Server

guzelcihad, 2018-12-01:
Hi,

I have using matrikon flex ua sdk server and freeopc ua client. I can connect server and get value of a node. But when i want to write a value to a variable in server, it gives me 'DataValue object has no attribute set_value' error messages. Can you help me?

Here is the python script:
[image]

brubbel, 2018-12-01:
set_value() is a wrapper method for setting the DataValue of a node instance.
So: myStoredVariable_node.set_value(...) should work.

guzelcihad, 2018-12-01:
Hi,

> set_value() is a wrapper method for setting the DataValue of a node object.
> So: `myStoredVariable_node.set_value(...)` should work.

closed 2018-12-01 by guzelcihad

guzelcihad, 2018-12-01:
When i try to look datavalue object type it prints:

> DataValue(Value:Variant(val:0,type:VariantType.UInt32), StatusCode:StatusCode(Good), SourceTimestamp:2018-12-01 17:45:54.496676)

After then, i try to write value like this:
`myStoredVariable.set_value(ua.Variant([23], ua.VariantType.UInt32))` or this :
`myStoredVariable_node.set_value(2)` .
But it gives me an error : 

> BadTypeMismatch: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)

Does it seem conflict?  I am giving a value with the same type of the datavalue object?
Is there somethinh that i miss?