FreeOpcUa/python-opcua #774: Server configuration: don't let enter anonymously

mikelga, 2018-12-10:
Hello!

I want to create a secure OPC-UA server on my Raspberry Pi using python-opcua. I want the OPC-UA client to have to enter mandatorily a Certificate and a Private Key to communicate with the server and with Basic256Sha256 + Sign&Encrypt security settings. For this, I used server-example.py as a reference program, and then, I made some changes:
server.load_certificate("my_certificate.der") 
server.load_private_key("my_certificate_key.pem") 
server.set_security_policy([
            ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
            ua.SecurityPolicyType.Basic256Sha256_Sign])
server.set_security_IDs(["Basic256Sha256"])
I thought that with these changes it would be enough, but when communicating with the server through the UaExpert client, I saw that I could communicate Anonymously, without having to use any client certificate. 

I try to connect Anonymously:
[image]

I accept the server certificate:
[image]

I am connected Anonymously. But, still, in the Server Diagnostics View appears that the SecurityMode is SignAndEncrypt and the PolicyId is certificate_basic256sha256. Why it let me connect Anonymously with this configuration? What more changes I have to do?
[image]

Thank you for the help!

oroulet, 2018-12-10:
The development focus has been on accepting as many clients as possible, not to be secure. Making it really secure will require to implement a lot of checks at different places. You are welcome to start the work. The architecture is supposed to support this and improvements are coming all the time

closed 2018-12-18 by mikelga

zerox1212, 2018-12-18:
We should probably write this on the front page README. A lot of people will probably see the certificate implementation and assume it's secure.