FreeOpcUa/python-opcua #777: [Server] method ConditionRefresh implementation

maxim-s-barabash, 2018-12-18:
opcua 0.98.5

Where to look to implement 'ConditionRefresh' method?
Server now returns BadNothingToDo
Аor test use example examples/client_to_prosys_events.py

zerox1212, 2018-12-18:
I'm not familiar with this this method, but example code would be helpful. Chances are the UA method isn't implemented.

maxim-s-barabash, 2018-12-18:
In which module are the UA methods implemented?

maxim-s-barabash, 2018-12-19:
I myself will try to implement this method.
But I do not find the right place in the code.

zerox1212, 2018-12-19:
I'm not familiar with this example. If you are using the python client to talk to Prosys and you get this error I think there is a problem with your server, not the client.

maxim-s-barabash, 2018-12-19:
server, also on python-opcua from examples

closed 2019-01-31 by oroulet