FreeOpcUa/python-opcua #785: Unknown message received FourByteNodeId(i=533)

JohannaSte, 2019-01-18:
Hello,

I am trying to exchange data between my OPC UA Server in Python and an OPC UA Client in Plant Simulation.

The connection is established and the server is connected. When trying to read the elements of the server using Plant Simulation, Python reports the following error in a continuous loop:
**Unknown message received FourByteNodeId(i=533)**

Why does this error occur?

My OPC UA Viewer can connect to the server and read all elements.

Thank you very much,

Johanna

oroulet, 2019-01-18:
it means the other part is sending us a message which we have not implemented. look at uaprocessor.py and you probably need to read spec to find out which one it is

zerox1212, 2019-01-18:
`BrowseNextRequest_Encoding_DefaultBinary = 533`

Need to look in spec what this is.

JohannaSte, 2019-01-21:
Thanks for the help. I'm also in contact with Plant Simulation Support. Let's see what happens.

peisenha91, 2020-03-18:
hi, i got the same Problem with a python opcua Server and Plant Simulation. Did you find a solution for the Problem? Thanks for your time in sdvance.

SoerenBergmann, 2020-03-19:
hi, i had the same problem, any solutions?

peisenha91, 2020-03-19:
hi, not yet. I hoped to get a response from JohannaSte.

zerox1212, 2020-03-19:
Sorry, I don't have time to look at the moment. Can one of you find this in the OPC UA specification?

peisenha91, 2020-03-19:
"BrowseNextRequest_Encoding_DefaultBinary = 533" this statement is as far as i get. What does it tell me?

zerox1212, 2020-03-19:
You have to register with the OPC foundation. Then you can download the OPC UA spec documents in PDF format. You have to find what `BrowseNextRequest_Encoding_DefaultBinary` is supposed to do.

peisenha91, 2020-03-20:
Thanks for the information. I will look it into it immediately

HSuser, 2020-08-24:
I have the exact same problem. @peisenha91 Did you find a solution for this problem?