FreeOpcUa/python-opcua #789: UAExpert error for every event and subscription

TryToLearn123, 2019-01-30:
Hi everyone,
I get an error in UAExpert every Time I get an event or subscription notification.
Then I checked the examples if the error also occurs there and it does.
The error seems to have no effect because everything works how it should.
Like it seems that everyone uses the UAExpert certainly someone else already recognized this error and maybe knows the reason for this error.

Here is an screenshot of the error:

[image]

zerox1212, 2019-01-30:
Is your server using the latest code from master branch?

oroulet, 2019-01-31:
I have never seen that... Keep it open

TryToLearn123, 2019-02-01:
I tried it on another system and there it works with no error messages.
So the error has something to do with my system.
Except of the error message everything works fine.
I close the issue.

closed 2019-02-01 by TryToLearn123