FreeOpcUa/python-opcua #791: trouble Calling get_child

RensRoodt, 2019-02-04:
hello guys,
I'm working on the Client-events.py, but I run into a few problems. 
The first problem is: obj = root.get_child(["0:Objects", "2:MyObject"]))
I have tried different variants to get it working:
obj= root.get_child(["0: Object, 2:p_HmiInterfacing, 2:Hmi_ArrDetectedSlaves"])
obj= root.get_child([" 2:p_HmiInterfacing, 2:Hmi_ArrDetectedSlaves"]))

In the OPCUA expert client is the node id: 
[image]
The browse path is:
[image]

when i run this code i get the same error: 
C:\Python34\python.exe "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.3\helpers\pydev\pydevd.py" --multiproc --qt-support=auto --client 127.0.0.1 --port 60435 --file C:/Users/rens.op.het.rood/Desktop/python-opcua-master/examples/client-events.py
pydev debugger: process 6880 is connecting

Connected to pydev debugger (build 183.5153.39)
Objects node is:  Node(TwoByteNodeId(i=84))
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.3\helpers\pydev\pydevd.py", line 1741, in <module>
    main()
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.3\helpers\pydev\pydevd.py", line 1735, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.3\helpers\pydev\pydevd.py", line 1135, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2018.3.3\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:/Users/rens.op.het.rood/Desktop/python-opcua-master/examples/client-events.py", line 85, in <module>
    obj = root.get_child(["0: Object, 2:p_HmiInterfacing, 2:Hmi_ArrDetectedSlaves"])
  File "..\opcua\common\node.py", line 467, in get_child
    result.StatusCode.check()
  File "..\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadNoMatch: The requested operation has no match to return.(BadNoMatch)

Process finished with exit code 1

I think this means that the browse path is wrong. But unfortunately I don't know how to do it? Can you give me a hint?

Furthermore I get stuck on :
myevent = root.get_child(["0:Types", "0:EventTypes", "0:BaseEventType", "2:MyFirstEvent"]))
I can't really find what I need to fill in here and what it does anywhere

zerox1212, 2019-02-04:
Your nodes are in namespace 4, not 2.

RensRoodt, 2019-02-06:
thx, i will close the issue!

closed 2019-02-06 by RensRoodt