FreeOpcUa/python-opcua #792: Problem with encryption from v0.98.5 to v0.98.6

robertofem, 2019-02-06:
Hello:

I have found a problem with encryption when upgrading from v0.98.5 to v0.98.6. The following client and server work in v0.98.5 but the client fails in v0.96.6. It looks a programming issue. The certificate and key have been generated with "generate_certificate.sh" file from the examples folder.

Regards

`from opcua import ua, Client

client = Client('opc.tcp://127.0.0.1:4840')

client.set_security_string("Basic128Rsa15,SignAndEncrypt,my_cert.der,my_private_key.pem")

client.connect()
while True:
    pass`

`from opcua import ua, Server

server= Server()

server.set_endpoint('opc.tcp://127.0.0.1:4840')

server.load_certificate("my_cert.der")
server.load_private_key("my_private_key.pem")
server.set_security_policy([ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt])

ns= server.register_namespace('http://www.website.com')

server.start()

while True:
    pass`

oroulet, 2019-02-06:
do you get an error?

oroulet, 2019-02-06:
the only commit related to security is that one: https://github.com/FreeOpcUa/python-opcua/commit/10131b76f69dc22bc533976f1e06b4809251ff18  as far as I could see

oroulet, 2019-02-06:
maybe there is an issue with depcrecated encryption... but I though we did not remove support for them

robertofem, 2019-02-06:
Sorry, I did not post the error: 

Traceback (most recent call last):
  File "client.py", line 5, in <module>
    client.set_security_string("Basic128Rsa15,SignAndEncrypt,my_cert.der,my_private_key.pem")
  File "/home/roberto/.virtualenvs/opcnew/lib/python3.6/site-packages/opcua/client/client.py", line 175, in set_security_string
    parts[4] if len(parts) >= 5 else None, mode)
  File "/home/roberto/.virtualenvs/opcnew/lib/python3.6/site-packages/opcua/client/client.py", line 187, in set_security
    endpoint = Client.find_endpoint(endpoints, mode, policy.URI)
  File "/home/roberto/.virtualenvs/opcnew/lib/python3.6/site-packages/opcua/client/client.py", line 141, in find_endpoint
    raise ua.UaError("No matching endpoints: {0}, {1}".format(security_mode, policy_uri))
opcua.ua.uaerrors._base.UaError: No matching endpoints: 3, http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15

robertofem, 2019-02-06:
I see in the commit that you implemented the new encryption. I will test the new (more secure) methods in 0.98.6. Thanks for the quick response!