FreeOpcUa/python-opcua #793: question: server and client example.py

jylilaht, 2019-02-10:
custom_nodes.xml tags are in namespace index 0 and other tags in namespace index 2. Is this intended behaviour somehow, I don't understand.

I am just trying to figure out how to ensure namespace index is right when I am getting tag value. Afaik I need to query namespace index of uri first then use that when I get tag

i.e. client.get_node("<here be namespace index i got from server>; s=mynodestring")

I am kind of new to this python opc-ua so maybe i just dont figure out something right.

oroulet, 2019-02-10:
I see the namespaces in these examples are a bit random.
but these two øines are correct, you should define custome namespace and use it

    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

on client side this should be 

idx = get_namespace_index(my_uri)

oroulet, 2019-02-10:
I just improved a litle the namsepace handling in these examples. They were outdated

jylilaht, 2019-02-11:
okay thanks a lot. this clears it up.

closed 2019-02-11 by jylilaht