FreeOpcUa/python-opcua #794: UaError: No future object found for request: 0, callbacks in list are [1]

pfc2003, 2019-02-11:
Hi,

I'm trying to connect to the Unified Automation Demo CPP Server, using Basic256Sha256 SignAndEncrypt.
The same error happens in my code and in your example "client_to_prosys_crypto.py, when I change it to match my server's url and certificate paths.

Unfortunately I'm not too deep into python and have no idea what this message means. 
Security Mode None works fine.

Any hints?

Regards

 

DEBUG:opcua.client.ua_client.Socket:Sending: CloseSecureChannelRequest(TypeId:FourByteNodeId(i=452), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-02-11 16:41:29.698260, RequestHandle:3, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)))
INFO:opcua.client.ua_client.Socket:Request to close socket received
INFO:opcua.client.ua_client.Socket:Socket closed, waiting for receiver thread to terminate...
INFO:opcua.client.ua_client.Socket:Socket has closed connection
INFO:opcua.client.ua_client.Socket:Thread ended
INFO:opcua.client.ua_client.Socket:Done closing socket: Receiving thread terminated, socket disconnected
INFO:opcua.client.ua_client.Socket:opening connection
INFO:opcua.client.ua_client.Socket:Thread started
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.uaprotocol:received header: Header(type:ACK, chunk_type:F, body_size:20, channel:0)
DEBUG:opcua.uaprotocol:Waiting for header
INFO:opcua.client.ua_client.Socket:open_secure_channel
DEBUG:opcua.client.ua_client.Socket:Sending: OpenSecureChannelRequest(TypeId:FourByteNodeId(i=446), RequestHeader:RequestHeader(AuthenticationToken:TwoByteNodeId(i=0), Timestamp:2019-02-11 16:41:29.714011, RequestHandle:1, ReturnDiagnostics:0, AuditEntryId:None, TimeoutHint:1000, AdditionalHeader:ExtensionObject(TypeId:TwoByteNodeId(i=0), Encoding:0, None bytes)), Parameters:OpenSecureChannelParameters(ClientProtocolVersion:0, RequestType:SecurityTokenRequestType.Issue, SecurityMode:MessageSecurityMode.Sign, ClientNonce:݇����":�0�*�{*��8Ǘ�%�e��9A��A, RequestedLifetime:3600000))
INFO:opcua.uaprotocol:received header: Header(type:ERR, chunk_type:F, body_size:8, channel:0)
WARNING:opcua.uaprotocol:Received an error: MessageAbort(error:StatusCode(Good), reason:None)
CRITICAL:opcua.client.ua_client.Socket:Received an error: MessageAbort(error:StatusCode(Good), reason:None)
ERROR:opcua.client.ua_client.Socket:Protocol Error
Traceback (most recent call last):
  File "/home/oem/.local/lib/python2.7/site-packages/opcua/client/ua_client.py", line 97, in _run
    self._receive()
  File "/home/oem/.local/lib/python2.7/site-packages/opcua/client/ua_client.py", line 115, in _receive
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/home/oem/.local/lib/python2.7/site-packages/opcua/client/ua_client.py", line 125, in _call_callback
    .format(request_id, self._callbackmap.keys())
UaError: No future object found for request: 0, callbacks in list are [1]

afcmrp, 2019-10-07:
Hello! Did you manage to solve this or get closer to a root cause? I'm facing a similar issue.

pfc2003, 2019-10-07:
Hi afcmpr, 
I could not not solve this and switched to a different OPC UA Stack...

afcmrp, 2019-10-07:
Ok, thanks for the speedy reply :)

pieAwd, 2022-11-11:
Hey @pfc2003, I have the same problem. Any chance you can share the solution you've switched to? Thanks in advance.