FreeOpcUa/python-opcua #795: Trouble getting display name from datachange_notification

bjosit, 2019-02-11:
Hi, 
Im building a project which takes values from a opc server, an inserts them into a SQL database. 
i have issues getting display name from datachange_notification.

     def datachange_notification(self, node, val, data):
         print(node.get_display_name.Text) 
Trying to get displayname

zerox1212, 2019-02-12:
You need to study up on python. `get_display_name()` is a function not an attribute.

`print(node.get_display_name())`

bjosit, 2019-02-12:
I know…. I can not use the node.get_display_name() because, i get an exception when we try to call up the class opc.ua.uatypes.LocalizedText inside datachange_notification function.

It looks like a naming error:

node.get_display_name is an class function bound  to the node class.

node.get_display_name() is an class instance from opc.ua.uatypes.LocalizedText.

TryToLearn123, 2019-02-12:
Try
`node.get_display_name().Text`

I think you tried to call the LocalizedText object

bjosit, 2019-02-12:
We have tried to call the LocalizedText object from the node object inside the datachange_notification function

def datachange_notification(self, node, val, data):
     print(node.get_display_name().Text) 

But we get an raise time error

zerox1212, 2019-02-12:
Post the error.

bjosit, 2019-02-13:
Code
##############
from opcua import Client

class Subhandler(object):

    def datachange_notification(self,node,val,data):
                                      
       print(node.get_display_name().Text)
    
if __name__ == "__main__":

    url = 'opc.tcp://192.168.11.102:4840'
    client = Client(url)
    client.connect()
   
    node = client.get_node('ns=3;s="dbOpc"."Random01"')
    handler = Subhandler()

    sub = client.create_subscription(1000,handler)
    sub_node = sub.subscribe_data_change(node)
   
   
error
#####################
Exception calling data change handler
Traceback (most recent call last):
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\opcua\common\subscription.py", line 135, in _call_datachange
    self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
  File "C:\Users\xxxxx\Desktop\Innlevering_1_DN\Innlevering_1_DN\module1.py", line 10, in datachange_notification
    print(node.get_display_name().Text)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\opcua\common\node.py", line 82, in get_display_name
    result = self.get_attribute(ua.AttributeIds.DisplayName)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\opcua\common\node.py", line 272, in get_attribute
    result = self.server.read(params)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\opcua\client\ua_client.py", line 326, in read
    data = self._uasocket.send_request(request)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\site-packages\opcua\client\ua_client.py", line 78, in send_request
    data = future.result(self.timeout)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64\lib\concurrent\futures\_base.py", line 434, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

oroulet, 2019-02-13:
You cannot do any network  call within the data change callback. This is documented everywhere possible ;-)

zerox1212, 2019-02-13:
Yes, what @oroulet said. Sorry I didn't notice that.

To explain, when you call `get_display_name()` you are actually trying to do another communication to the server from the network thread and it's blocking.

You will have to set an internal flag, then from you main thread do the `get_display_name()` call on all nodes that are "marked".

cyberRobotic, 2019-02-13:
who to start a thread inside the data_change_notification more than once ?

zerox1212, 2019-02-13:
With the current design I would focus on saving whatever datachange data you need to your own python object/variable. Then operate on the new data somewhere else.

bjosit, 2019-02-13:
Allright thx

closed 2019-02-13 by bjosit

zerox1212, 2019-02-13:
@oroulet We have talked about this before. Would it be good to use `Queue` with the `datachange` call back so it's possible to execute functions on the main thread?

I still feel like we need an improvement here. People keep needed to write boiler plate code just to interact with the new data.