FreeOpcUa/python-opcua #796: How to get All tags name or xml file in client from the server

MickyP24, 2019-02-12:
Hi,
I am new in the python as well as OPC UA, as I am trying to developed a system in which client can import all the tag name from the server (example: abc.simulator.random.value). I am using Prosys OPC UA server. 
While working on the library Started with example code and able to read node data if I know the tag node/child.
here is my code 
print("client connect")
root = client.get_root_node()
print("root:", root)
objects = client.get_objects_node()
print(objects)
childnode = objects.get_child("5:Simulation")
print(childnode)
while True:
   counter = client.get_node("ns=5;s=Counter1")
   count = counter.get_value()
   print(count)
   time.sleep(1)

Now I want to probe server for tag list (same as in milo) and that tag list push to the UI for selecting required tags so that I can scanned only selected tags.
Is this possible by using this library? I found get_export option in document but I am not cleared how to use it
Thanks in Advance for you reply.

zerox1212, 2019-02-12:
The client part of this library just gives you the OPC UA tools to interact with an OPC UA server. From there you can do whatever you want.

You will need to understand object oriented programming and python basics. Then you will need to write code that traverses your server and stores all the nodes you are interested. From there you will need to let your UI configure subscriptions to the selected node.

You can probably look at https://github.com/FreeOpcUa/opcua-client-gui for ideas, but this isn't something you will learn to do in a few hours if this is your first time with OPC UA and Python.

MickyP24, 2019-02-13:
Thanks zerox1212,
I am working on it, After going through a module code document I am able to get value, timestamp, datatypes and status code. Also I am trying implement logic to get the all tag names from the server with reference node id. 
Thanks for explaining.

oroulet, 2019-02-13:
Btw  there is nothing called "tag" in opcua. If you mean 'node' then asking the server for so modes is probably a bad idea and not necessary

MickyP24, 2019-02-15:
@zerox1212 , How I get Node id of the child. like server> xxx>yyy>zzz> sensorData. If I want to find sensor Data node ID . I tried with get_reference, get_children_description but all it received in list. If I want to get nodeID for "sendorData" then is their any command/function? 

One more update I used recursion logic to get all node name.
Thanks in Advance!!

zerox1212, 2019-02-15:
If you want to get specific nodes you can use the full path with `get_child()`. If you don't know the path use the Unified Automation Ua Expert test client to browse until you find it.

Something like this:
`root = client.get_root_node()
sensor = root.get_child( ["0:Objects", "3:xxx", "3:yyy", "3:zzz", "3:sensorData"] )`

MickyP24, 2019-03-07:
Hi Guys, 
I found the solution and able to received OPC data from Prosys OPC server.
Thanks for help @zerox1212 @oroulet

closed 2019-03-07 by MickyP24

amisha218, 2019-06-07:
@MickyP24 hey!! Can you please share your recursion logic

zerox1212, 2019-06-07:
Please take the time to try something yourself before posting in issues. Calling `node.get_children()` multiple times to find all nodes is not a difficult thing to do.

amisha218, 2019-06-07:
I know that....node.get_children() will give the node names but I want to get the values of those nodes for which node.get_value is generating an error.

zerox1212, 2019-06-07:
Your above post says nothing about `node.get_value` having an error. This issue is about listing all nodes, not getting values.

amisha218, 2019-06-10:
Then how doI will get the values of all the tags/nodes specified