FreeOpcUa/python-opcua #800: Frequent exceptions BadSessionNotActivated and BadSessionIdInvalid

ap-0, 2019-02-21:
Hello, 

I have written an process that creates multiple threads. Each thread connects to  the same OPC UA server and subscribes to various number of nodes (I mainly do this because the nodes change very frequently and since the OPC UA client is not thread safe I use multiple clients with multiple sockets). Most of the time the process is running finge. But from time to time I run into an `BadSessionNotActivated ` or `BadSessionIdInvalid ` exception when reading a nodes value.

Does the server have a session timeout? But why do I run into this timeout? I'm constantly reading data. Only once daily there is a 2 hour period where no nodes could change. 

My first question would be what these exceptions mean and why I run into them. The OPC UA documentation isn't very helpful.

oroulet, 2019-02-21:
Not easy to know.. There is a thread running in Client that renew the session a long time before it goes out, but since everythin is async, then maybe there is a race condition and we send or receive a message with old ID... 
To debug this you will probably have to print all Session Ids in a log together with renew time and see if something strange is happening...
you can also use wireshark or put logging on, but you will have so much output that you will probably never understand what is happening..

ap-0, 2019-09-23:
I am still facing this issue. After some time running fine my application looses suddenly connection to the OPC-UA server while reading data from a node with a `BadSessionNotActivated` exception. 

 ```
File "/opt/(...)/lib/python3.4/site-packages/opcua/common/node.py", line 129, in get_value
   result = self.get_data_value()
 File "/opt/(...)/lib/python3.4/site-packages/opcua/common/node.py", line 138, in get_data_value
   return self.get_attribute(ua.AttributeIds.Value)
 File "/opt/(...)/lib/python3.4/site-packages/opcua/common/node.py", line 244, in get_attribute
   result = self.server.read(params)
 File "/opt/(...)/lib/python3.4/site-packages/opcua/client/ua_client.py", line 284, in read
   response.ResponseHeader.ServiceResult.check()
 File "/opt/(...)/lib/python3.4/site-packages/opcua/ua/uatypes.py", line 201, in check
   raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionNotActivated: The session cannot be used because ActivateSession has not been called.(BadSessionNotActivated)
```

My first intention in these cases was to close the existing connection and create a new one to the server. But this does not help because I then run into a `TimeoutError`.

```
File "/opt/(...)/lib/python3.4/site-packages/opcua/client/client.py", line 224, in connect
  self.send_hello()
File "/opt/(...)/lib/python3.4/site-packages/opcua/client/client.py", line 251, in send_hello
  ack = self.uaclient.send_hello(self.server_url.geturl())
File "/opt/(...)/lib/python3.4/site-packages/opcua/client/ua_client.py", line 219, in send_hello
  return self._uasocket.send_hello(url)
File "/opt/(...)/lib/python3.4/site-packages/opcua/client/ua_client.py", line 154, in send_hello
  ack = future.result(self.timeout)
File "/opt/python34/lib/python3.4/concurrent/futures/_base.py", line 404, in result
  raise TimeoutError()
concurrent.futures._base.TimeoutError
```

The only thing that seems to help from here on is to stop the applicaiton and start it again but why if the connection is createdy newly?

oroulet, 2019-09-23:
Calling connect again is the same as restart do maybe add a sleep. But why you get that error is hard to say... I suppose this is the session timeout. Do you use the latest version? There has been a bug there

ap-0, 2019-09-23:
I am calling sleep(10) between every connection attempt. Every time the connection failes I create a new Client for the next attempt. 

How the session times out is not clear to me because the frequency I received subsription notifications and read data nodes is high. Couple of seconds only. 

I am using version a rather old verison 0.90.0.

oroulet, 2019-09-23:
Then you need to try to upgrade