FreeOpcUa/python-opcua #803: Error: Number of acknowledgement results (0) does not match number of acknowledgements(1)

Cesaario, 2019-02-26:
I'm new to freeopcua and I'm getting an error that I can't get rid of.
My server works fine, and I can read the values at UaExpert, but I'm recieving several "Error: Number of acknowledgement results (0) does not match number of acknowledgements(1)"
This is my server:
`from opcua import Server
from random import randint
import time
import sys

server = Server()

server.set_endpoint("opc.tcp://localhost:4840")

addspace = server.register_namespace("OPCUA")

node = server.get_objects_node()

Param = node.add_object(addspace, "Parametros")

Temp = Param.add_variable(addspace, "Temp", 0)

Temp.set_writable()

server.start()
print("Server iniciado...")

while True:
	try:
		val_temp = randint(0,100)
		Temp.set_value(val_temp)
		print(val_temp)
		time.sleep(5)
	except KeyboardInterrupt:
		print("Finalizando servidor...")
		break

server.stop()`

Should I be worried about this error?

oroulet, 2019-02-26:
Not sure. What client are you using?

Cesaario, 2019-02-27:
Not sure. I'm using Unified Automation UaExpert.

TryToLearn123, 2019-02-28:
I had/have the same error. On one of my PCs I get these errors and on an other PC I didn´t get the errors with the same server.
It has no impact on the functionality so you shouldn care about it I think.

EEflow, 2019-06-26:
I see the same error using Unified Automation UaExpert 1.5.1.
I don't  notice any other negative effects but it would be nice to know what causes the error.

dannmartens, 2019-08-02:
I ended up here for the same reason, also using Unified Automation UaExpert 1.5.1. 
[image]

oroulet, 2019-08-02:
I do not get that error with uaexpert 1.4..... can it be something new with uaexpert 1.5.1?

oroulet, 2019-08-02:
I do not get the error using MySin variable in server-example.py dragged into "DataAccess view" in uaexpert . Should I do something else to reproduce this?

EEflow, 2019-08-02:
> I do not get the error using MySin variable in server-example.py dragged into "DataAccess view" in uaexpert . Should I do something else to reproduce this?

Nothing else is needed, subscription to MySin var should produce the error:
[image]

I can't test it with an older uaxport version, since these are not available for download anymore.

oroulet, 2019-08-02:
I cannot reproduce with 1.5.1 either... Do you use master from GitHub?

EEflow, 2019-08-02:
I have used pip to install this version:
```
opcua                     0.98.7                   pypi_0    pypi
```

oroulet, 2019-08-03:
Can you try master then. I kind of remember some changes there...

dannmartens, 2019-08-03:
I am currently building against HEAD of master: I do see that error in UaExpert.

maxim-s-barabash, 2019-08-28:
I also see this message.
opcua                     0.98.7  
UaExpert                1.5.0

There are nodes in the tree with EventNotifier.
[image]

ralphswork, 2019-08-28:
I also see the message in UaExpert 1.5.1. 
With opcua 0.98.7 as well as with asyncua 0.5.1

yilmaznaslan, 2019-12-03:
Same Error

brubbel, 2020-03-15:
Error message is caused by missing PublishResult.Results acknowledgements.

**Background:**
1. client subscribes to one or more datachanges.
2. client sends publication request to server.
3. datachange event: server sends publication response, with incremental "notification id"
4. client sends new publication request to server, acks each "notification id"
5. datachange event: publication response + **statuscode of processing for each above ack**
This is the missing part in freeopcua at this moment.

The OPCUA spec is somewhat vague about the implementation at this point. From what I understand, the only way this ack/confirm procedure can work is when the client sends a publication request after each publication response. This request can then be used as "token" (ie increment notification id) for the next publication response.

The way freeopcua works at the moment is that datachanges are published whenever data changes (throttled and multiple changes are bundeled at regular intervals), however this implies that e.g. 2 datachange publications responses are scheduled for tx to the client while the client expects a sequential PublishResult.Results for each ack in the publish request, **in the order they were issued by the client** (flaw in the OPCUA spec imho).

I did some testing and this stops the errors in uaexpert, however it has considerable impact on the way freeopcua currently implements datachange events. Also, this server-client-server roundtrip basically kills concurrency and fast updates.
On the other hand, ignoring this floods the UaExpert log with Error messages

aschaepper, 2020-09-14:
Hi together

I got the same error here, however I understand from you answer that its not a quickfix. I just want to add that the "Prosys OPC UA Client" does not log errormessages, but this client does not update the opc ua variables when they change. So it can be, depending on the implementation of the client that its not properly working.

With UA Expert, I just get the message as in the thread's titel, but the value is refreshed.

So depending on the client implementation this error can be more severe.

yichenj, 2021-02-08:
Hi, is there any updates for this issue? Will it be fixed, or any walk around?