FreeOpcUa/python-opcua #804: Method to return an Array

TPG4, 2019-02-26:
Hello,

I'm trying to call a Matlab function from a Beckhoff PLC with the OPC-UA client (for testing I use UAExpert)  and for this reason, I use the FreeOpcUa Server, because there is no Matlab OPC-UA Server from Matlab but there is the Matlab engine for Python. So to say, the FreeOpcUa is the "Middleware" in-between. On the other hand, rewriting the code in Python is not possible because of the used functionalities of the optimization package of Matlab, the effort would be huge. 
I first tried to call a simple function with 2 input parameters which return the [triangle area](https://uk.mathworks.com/help/matlab/matlab_prog/scripts-and-functions.html?searchHighlight=triarea&s_tid=doc_srchtitle), which works fine. This is almost the same method with the "multiply" in [server-method.py](https://github.com/TPG4/python-opcua/blob/master/examples/server-methods.py).

Then i tried to call to call the optimization function which gets 4 Double input arguments and returns an array (or one dim. Matrix) with 24 Double values. Here is the code:  

`import sys
sys.path.insert(0, "..")
import logging
import time
import matlab
import matlab.engine
#import numpy
#from enum import Enum, IntEnum

try:
    from IPython import embed
except ImportError:
    import code

    def embed():
        vars = globals()
        vars.update(locals())
        shell = code.InteractiveConsole(vars)
        shell.interact()

from opcua import ua, uamethod, Server
# method to be exposed through server
def func(parent, variant):
    ret = False
    if variant.Value % 2 == 0:
        ret = True
    return [ua.Variant(ret, ua.VariantType.Boolean)]

# method to be exposed through server
# uses a decorator to automatically convert to and from variants

@uamethod
def schedule(parent, Pel_max,storageCapacity, SOC, pl_min):
    print("Schedule Input parameter : ", Pel_max,storageCapacity, SOC, pl_min)
    eng = matlab.engine.start_matlab()
    ret = eng.schedule(Pel_max,storageCapacity, SOC, pl_min)
    print(ret)
    return ret

if __name__ == "__main__":
    # optional: setup logging
    logging.basicConfig(level=logging.WARN)
    #logger = logging.getLogger("opcua.address_space")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.internal_server")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.binary_server_asyncio")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.uaprocessor")
    # logger.setLevel(logging.DEBUG)
    #logger = logging.getLogger("opcua.subscription_service")
    # logger.setLevel(logging.DEBUG)

    # now setup our server
    server = Server()
    #server.set_endpoint("opc.tcp://localhost:4840/freeopcua/server/")
    server.set_endpoint("opc.tcp://192.168.1.12:4848/freeopcua/server/")
    server.set_server_name("FreeOpcUa Example Server")

    # setup our own namespace
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

[... 85 lines cut ...]
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 253, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 293, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 293, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 271, in to_binary
    return pack_uatype(vtype, val)
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 189, in pack_uatype
    return variant_to_binary(value)
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 375, in variant_to_binary
    b.append(pack_uatype(var.VariantType, var.Value))
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 185, in pack_uatype
    return extensionobject_to_binary(value)
  File "C:\ProgramData\Anaconda3\envs\py36\lib\site-packages\opcua\ua\ua_binary.py", line 455, in extensionobject_to_binary
    TypeId = ua.extension_object_ids[obj.__class__.__name__]
KeyError: 'double'

As you can see the **schedule** function returns the array for the given inputs, but in UAExpert I don't get the array as an output Argument, with an Error: Failed with error 'BadTimeout'
[image]

I suppose the problem occurs because of the Datatype mismatch (see ua_binary.py), because I declare the output Argument as a Double and not as an array (or List). 
Could you please help me with this issue and advice me how to declare the function output argument as an array of Double? OR is there any workaround for that?

Thank you in advance!

zerox1212, 2019-02-26:
You are returning a list with array inside it. Out of curiosity does it work if you do `return ret[0]`

TPG4, 2019-02-27:
Hello Andrew,
Thanks for the quick responce! 
I tried with `return ret[0]` but it won't work as well. 

@oroulet and  @destogl have addressed this issue in #35:
```
ValueRank:
Indicates whether the dataType is an array and how many dimensions the array has.
It may have the following values:
n > 1: the dataType is an array with the specified number of dimensions.
OneDimension (1): The dataType is an array with one dimension.
OneOrMoreDimensions (0): The dataType is an array with one or more dimensions.
Scalar (−1): The dataType is not an array.
Any (−2): The dataType can be a scalar or an array with any number of dimensions.
ScalarOrOneDimension (−3): The dataType can be a scalar or a one dimensional array.
NOTE All DataTypes are considered to be scalar, even if they have array-like semantics like ByteString and String.
```
In my case the ValueRank should be (0) or (-2) or (-3), but I still get the same error.
I tried all values for ValueRank with and without ArrayDimensions, but still get the errors.

zerox1212, 2019-02-27:
Can you try making a static OPC UA variable that is an array first? That way we can find out if the problem is specifically with methods. Unfortunately I never used arrays when I worked with OPC UA.

TPG4, 2019-03-19:
It works with a static array, the method returns the array as expected. 
The problem was that I was getting a list with 24 lists([24x1]), so I had to rewrite the outcome in an array that would be returned from the method.

closed 2019-03-19 by TPG4