FreeOpcUa/python-opcua #805: Embed function with subscribtion

RensRoodt, 2019-02-28:
Hello all,

I am making subscribties and that works fine, but I also have a GUI running pyqt4. I know there is a GUI with OPC UA however it uses pyqt5 and the company does not want to upgrade.

If there is a datachange, a function is called that will do all kinds of things, but the moment the subscriber is connected, the GUI freezes. 
This is because of the function embed. But if we remove it, the next error will occur:
opcua.ua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client (BadSessionClosed). 
my question to you is actually why is the embed() used, and is it possible to get it "easily" Gui?

With kind regards 
Rens

oroulet, 2019-02-28:
If you want to make your own GUI you should probably look at code in the qt5 GUI. The embed() is for debugging, just remove it , it is about the same as pydb.set_trace()

RensRoodt, 2019-02-28:
Thanks for your reaction, I've tried it it works now!

closed 2019-02-28 by RensRoodt