FreeOpcUa/python-opcua #807: 'NoneType' object has no attribute 'socket' with Example/client_to_kepware.py

MickyP24, 2019-03-06:


oroulet, 2019-03-06:
You did not connect or did not start server

MickyP24, 2019-03-06:
@oroulet, Started Kepware server as shown below
[image]

MickyP24, 2019-03-06:
Again reinitialize I got this exception,
Exception has occurred: opcua.ua.uaerrors._auto.BadSessionNotActivated
The session cannot be used because ActivateSession has not been called.(BadSessionNotActivated)
  File "C:\Users\mpaithankar\Downloads\python-opcua-master\python-opcua-master\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
  File "C:\Users\mpaithankar\Downloads\python-opcua-master\python-opcua-master\opcua\client\ua_client.py", line 296, in close_session
    pass
  File "C:\Users\mpaithankar\Downloads\python-opcua-master\python-opcua-master\opcua\client\client.py", line 474, in close_session
    return self.uaclient.close_session(True)
  File "C:\Users\mpaithankar\Downloads\python-opcua-master\python-opcua-master\opcua\client\client.py", line 275, in disconnect
    self.disconnect_socket()
  File "C:\Users\mpaithankar\Downloads\python-opcua-master\python-opcua-master\examples\client_to_kepware.py", line 53, in <module>
    client.disconnect()

zerox1212, 2019-03-06:
All firewall stuff disabled?

MickyP24, 2019-03-07:
Thanks zerox1212 for reply..
Yes... Firewall is disabled. 
So I am using visual code studio on windows 10 and KepwareserverEx is also installed on same machine.
I check this with prosys OPC server it doesn't give me any issue. But now My requirement is to work with  kepware and  is not working.

zerox1212, 2019-03-07:
If it works with Prosys then you probably need to take a look at the Kepware server settings or other issues. Kepware OPC UA implementation is notoriously bad.

Will UA Expert connect to your Kepware server?

MickyP24, 2019-03-08:
Yes @zerox1212 , You are right its create an error in UA Export
[image]

How can I solve this?

MickyP24, 2019-03-08:
On UA Export, It ask me a username, password to connect
If added default as user then it directly disconnect from client
[image]

zerox1212, 2019-03-08:
Seems like your Kepware server isn't allowing any connections.

MickyP24, 2019-03-11:
I think So...