FreeOpcUa/python-opcua #811: [Client] activate session with security policy Basic256Sha256 fails

mihainstr, 2019-03-20:
https://github.com/FreeOpcUa/python-opcua/blob/cdcaed36b772abf0ea1dffe42b2d8cdaada8f5e1/opcua/client/client.py#L414

**ISSUE:**

Attempting a CLIENT connect with the security policy set to Basic256Sha256 will fail when trying to activate the session. 

The issue is related to the incorrect ClientSignature.Algorithm used as a parameter for the activate session request, the current implementation uses "http://www.w3.org/2000/09/xmldsig#rsa-sha1" as the default algorithm which corresponds to  "http://opcfoundation.org/UA/SecurityPolicy/#Basic256". 

**FIX:**
Changing the ClientSignature.Algorithm to "http://www.w3.org/2000/09/xmldsig#rsa-sha256" activates the session correctly when security policy is set to Basic256Sha256.

The permanent fix should ensure that the correct encryption algorithm is deduced based on the used security policy.

sixdiamants, 2019-08-10:
I faced the same issue running the python client against a [UA-.NETStandard ](https://github.com/OPCFoundation/UA-.NETStandard)server.  The server logged this:

> 26-7-2019 09:51:09.930 Server - SESSION ACTIVATE failed. Unexpected signature algorithm for Basic256Sha256/Aes128_Sha256_RsaOaep: http://www.w3.org/2000/09/xmldsig#rsa-sha1
Expected signature algorithm: http://www.w3.org/2001/04/xmldsig-more#rsa-sha256  

as reported [here](https://opcfoundation.org/forum/opc-ua-implementation-stacks-tools-and-samples/sha2-signed-certificate-is-interpreted-as-sha1-signed-and-wrongly-rejected/).

Applying the fix mentioned above by @mihainstr still returned an unexpected signature algorithm. Only when I replaced 
 >  params.ClientSignature.Algorithm = "http://www.w3.org/2000/09/xmldsig#rsa-sha1"   

by the URI suggested by the server log
 
>params.ClientSignature.Algorithm =  "http://www.w3.org/2001/04/xmldsig-more#rsa-sha256"  

Did the server accept the connection. In other words, the URI to the algorithm for the signature method `http://www.w3.org/2000/09/xmldsig#rsa-sha256` didn't work for me.

alexanderpetrenz, 2020-04-14:
Is it desired to make this algorithm string configurable?

oroulet, 2020-04-14:
It should be. But where does it make sense?

floriangraetz, 2020-06-24:
The commit from oroulet seems to circumvent this precise issue, when SHA256 is used. However, this setting should be configurable, because other servers may not want to use SHA256. Shouldn't this setting be derived from the security string?

ZuZuD, 2020-07-01:
I use Basic256Sha256 with SignAndEncrypt security mode and this change broke my authentication. 

It seems my server only supports rsa-sha1 for the UserTokenSignature.Algorithm and I end up with ```params.UserTokenSignature.Algorithm != params.ClientSignature.Algorithm``` in my ```ActivateSessionRequest```.

I looked in the specifications but haven't found anything that stipulates the above condition doesn't align with it.

Part 4.7.36.5:
> This token shall always be accompanied by a Signature in the userTokenSignature parameter of ActivateSession if required by the SecurityPolicy. The Server should specify a SecurityPolicy for the UserTokenPolicy if the SecureChannel has a SecurityPolicy of None.

Part 4.5.6.3.2:
> userTokenSignature | SignatureData | If the Client specified a user identity token that supports digital signatures, then it shall create a signature and pass it as this parameter. Otherwise the parameter is null. The SignatureAlgorithm depends on the identity token type.

If I haven't missed anything, we should then probably let the option to pick the right algorithm rather than it being derived from the security string.

Also since #944 the request's Algorithm is "sha256" but we still sign with sha1 and this seems wrong to me: https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/client.py#L476-L479

oroulet, 2020-07-01:
merge request welcome. I do not use encryption so I do not follow this subject closely