FreeOpcUa/python-opcua #812: Could not connect to server: BadCommunicationError

Cesaario, 2019-03-25:
Hi, I'm running an OPC Server in a computer and trying to read the values from it from another computer. They are connected to the same network and the server computer has it's firewall disabled.
Here is the server code:
It just reads the value from a json string recieved from the serial port and then perform an action, that could be creating a node, updating it, etc. The important part is just until the server is started.
```
import serial
import json
import opcua
from opcua import Server

ser = serial.Serial("COM6", 115200)

server = Server()
server.set_endpoint("opc.tcp://localhost:4840")
server.start()

addspace = server.register_namespace("ESP32")
node = server.get_objects_node()
var = node.add_object(addspace, "Variaveis")

def handleRequest(data):
    tipo = data['tipo']
    if(tipo == 'declarar'):
        nome = data['nome']
        nodeid = "ns=2;s=" + nome
        print("declarando variavel:", nome)
        var.add_variable(nodeid, nome, 0).set_writable()
    if(tipo == 'atualizar'):
        variavel = data['nome']
        valor = data['valor']
        nodeid = "ns=2;s=" + variavel
        print("atualizando " + variavel + " para", valor)
        server.get_node(nodeid).set_value(valor)

while(1):
    if(ser.in_waiting > 0):
        leitura = ser.read_until(b'}').decode('ascii')
        leitura = leitura[leitura.rfind("{"):]
        data = json.loads(leitura)
        handleRequest(data)
```

Here is the client I'm trying to use to connect
```
from opcua import Client
import time

client = Client("opc.tpc://192.168.15.17:4840/freeopcua/server")

client.connect()
print("Cliente conectado")

while True:
	try:
		Temp = client.get_node("ns=2;i=2")
		val_temp = Temp.get_value()
		print(val_temp)
		time.sleep(2)
	except KeyboardInterrupt:
		break

client.disconnect()
```

But I recieved this error:

> Traceback (most recent call last):
  File "client.py", line 6, in <module>
    client.connect()
  File "C:\Users\Gustavo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\opcua\client\client.py", line 255, in connect
    self.connect_socket()
  File "C:\Users\Gustavo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\opcua\client\client.py", line 280, in connect_socket
    self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)
  File "C:\Users\Gustavo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\opcua\client\ua_client.py", line 245, in connect_socket
    return self._uasocket.connect_socket(host, port)
  File "C:\Users\Gustavo\AppData\Local\Programs\Python\Python37-32\lib\site-packages\opcua\client\ua_client.py", line 142, in connect_socket
    sock = socket.create_connection((host, port), timeout=self.timeout)
  File "C:\Users\Gustavo\AppData\Local\Programs\Python\Python37-32\lib\socket.py", line 727, in create_connection
    raise err
  File "C:\Users\Gustavo\AppData\Local\Programs\Python\Python37-32\lib\socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [WinError 10061] Nenhuma conexão pôde ser feita porque a máquina de destino as recusou ativamente (No connection could be made because the target machine actively refused them)

And if I try to connect from UaExpertClient using the same endpoint (opc.tpc://192.168.15.17:4840/freeopcua/server) i get:
[https://prnt.sc/n2p67t](url)

I already checked the IP and disabled the firewall from the computer server.
Is there any solution to this error?

unista85, 2019-10-23:
Hello Cesaario,

I have a similar problem.
Have you find any solution to your problem ?

Best regards.

zerox1212, 2019-10-23:
Are you on a corporate network?

oroulet, 2019-10-23:
Yes this is ip setup/firewall related. Try pinging and network scanners, etc.

unista85, 2019-10-24:
It works when server and client are on my computer but doesn't work when I move the server in an other computer.
Only client is based on Freeopcua.
I tried to disable the firewall but result is the same