FreeOpcUa/python-opcua #816: set_value to node

danSC90, 2019-03-29:
as a client i try to set the value of a node on my server like this:

from opcua import Client

url = "opc.tcp://10.0.0.55:4840"
client = Client(url)
client.connect()
Test = client.get_node("n=2;i=5")
Test.set_value(10.0)

On the server "Test" is already set as writeable.

But i always get this error:
Traceback (most recent call last):
  File "/Users/d_naschberger/Documents/workspace/OPCUAclient/client.py", line 25, in <module>
    Test.set_value(10.0)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/opcua/common/node.py", line 214, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/opcua/common/node.py", line 259, in set_attribute
    result = self.server.write(params)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/opcua/client/ua_client.py", line 346, in write
    data = self._uasocket.send_request(request)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/opcua/client/ua_client.py", line 78, in send_request
    data = future.result(self.timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/concurrent/futures/_base.py", line 434, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

somebody may know whats happening here?

thx cheers

zerox1212, 2019-03-29:
What server are you using?

danSC90, 2019-03-29:
i wrote a server using opcua like this:

```
from opcua import Server

server = Server()

url = "opc.tcp://10.0.0.55:4840"
server.set_endpoint(url)

name = "OPCUA_SERVER"
addspace = server.register_namespace(name)

node = server.get_objects_node()

Param = node.add_object(addspace, "Parameters")

Test = Param.add_variable(addspace, "Test", 0)

Test.set_writable()

server.start()
```

thanks!

Muzhai, 2020-11-12:
I encountered the same problem. Have you found the cause of the problem?

AndreasHeine, 2020-11-12:
reading and writing nodes works! so please provide more details code snippets and what server type!? variable type please make sure if you create variables to provide a propper variant format otherwise it can make problems because python is dynamicly typed and opcua is strict typed! lg

Muzhai, 2020-11-12:
I used the “python-opcua/examples/server-minimal.py” as a test Server. This is working gut and the function set_value is no problem too. But when i try to write certain values through "client-minimal.py" occurred the Error "concurrent.futures._base.TimeoutError".

The Client lick this
import sys
sys.path.insert(0, "..")

import time
from opcua import ua, Client

if __name__ == "__main__":

    client = Client("opc.tcp://localhost:4845/freeopcua/server/")
    # client = Client("opc.tcp://admin@localhost:4840/freeopcua/server/") #connect using a user
    try:
        client.connect()

        # Client has a few methods to get proxy to UA nodes that should always be in address space such as Root or Objects
        root = client.get_root_node()
        print("Root node is: ", root)

        # Node objects have methods to read and write node attributes as well as browse or populate address space
        print("Children of root are: ", root.get_children())

        # get a specific node knowing its node id
        var = client.get_node(ua.NodeId(1, 2))
        print(var)
        # var = client.get_node("ns=2;i=1")
        #print('value', var.get_data_value())# get value of node as a DataValue object
        #var.get_value() # get value of node as a python builtin
        #var.set_value(ua.Variant([50], ua.VariantType.Double)) #set node value using explicit data type
        var.set_value(11) # set node value using implicit data type

        # Now getting a variable node using its browse path
        myvar = root.get_child(["0:Objects", "2:MyObject", "2:MyVariable"])
        obj = root.get_child(["0:Objects", "2:MyObject"])
        print("myvar is: ", myvar)
        print("myobj is: ", obj)

        # Stacked myvar access
        print("myvar is: ", root.get_children()[0].get_children()[1].get_variables()[0].get_value())

    finally:
        client.disconnect()

AndreasHeine, 2020-11-12:
could you tell me which one causes that? so i can try to reproduce?

Muzhai, 2020-11-12:
This is the Server

import sys
sys.path.insert(0, "..")
import time

from opcua import ua, Server
server = Server()
server.set_endpoint("opc.tcp://0.0.0.0:4845/freeopcua/server/")
objects = server.get_objects_node()
uri = "http://examples.freeopcua.github.io"
idx = server.register_namespace(uri)
# populating our address space
myobj = objects.add_object(idx, "MyObject")
myvar = myobj.add_variable(idx, "MyVariable", 6.7)
myvar.set_writable()  # Set MyVariable to be writable by clients
print(myvar)
# starting!
server.start()
try:
    count = 0
    while True:
        time.sleep(1)
        count += 0.1
        myvar.set_value(count)
finally:
    # close connection, remove subcsriptions, etc
    server.stop()

Muzhai, 2020-11-12:
This is the Error:
Traceback (most recent call last):
  File "D:/Masterarbeit/program/MA_OPC_UA/OPC_UA_Client.py", line 31, in <module>
    var.set_value(11) # set node value using implicit data type
  File "D:\Programm\Python\lib\site-packages\opcua\common\node.py", line 217, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "D:\Programm\Python\lib\site-packages\opcua\common\node.py", line 262, in set_attribute
    result = self.server.write(params)
  File "D:\Programm\Python\lib\site-packages\opcua\client\ua_client.py", line 367, in write
    data = self._uasocket.send_request(request)
  File "D:\Programm\Python\lib\site-packages\opcua\client\ua_client.py", line 83, in send_request
    data = future.result(self.timeout)
  File "D:\Programm\Python\lib\concurrent\futures\_base.py", line 434, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

Muzhai, 2020-11-12:
Fist of all thank you for your help , I am a newbie, recently I am work with OPCUA and that makes me excited and depressed:)

AndreasHeine, 2020-11-12:
var = client.get_node(ua.NodeId(1, 2)) <--- is type object and not variable you can only write variables!

you should use uaexpert to check the ids befor try to write them 
var = client.get_node(ua.NodeId(2, 2))

to simplify you can `client.get_node("ns=2;i=2")`

[image]
[image]
[image]

Muzhai, 2020-11-12:
Thank you very much, the problem has been resolved. :)))