FreeOpcUa/python-opcua #818: Set value to tag

lcflorindo, 2019-04-09:
Hi all.
I want to write '1' and '0' in two tags ('peca_NOK' and 'peca_OK') from my kepware:
`import opcua
from opcua import Client, ua

client = Client("opc.tcp://127.0.0.1:49320")

root = client.get_root_node()
node = client.get_node("ns=2;s=CAMERA.WS.peca_NOK")
tag1 = node.get_data_value()`

but still getting error:
`AttributeError: 'NoneType' object has no attribute 'send_request'`

someone can give me a light?

zerox1212, 2019-04-09:
Can you write a value to this node on UA Expert?

oroulet, 2019-04-09:
You did not call connect ()

lcflorindo, 2019-04-09:
> Can you write a value to this node on UA Expert?

Didn't tried on it actually, idk for sure how to it, I just started in OPC-UA. The kepware was build by other technician guy.

> You did not call connect ()

if i call, i get this error:
` gaierror: [Errno -3] Temporary failure in name resolution `

oroulet, 2019-04-10:
So connect. Does not succeed and you expect something to with after?!?

oroulet, 2019-04-10:
Read the error message. It says that the name of the machine is wrong. You have the wrong name or wrong ip or bad DNS setup

lcflorindo, 2019-04-10:
> So connect. Does not succeed and you expect something to work after?!?

My bad. I've fixed the IP now i can connect normally. Now re-writting the code using some sources that I found, even converting type of values I'm getting:
` node.set_attribute(ua.AttributeIds.Value, 1)`
Return:
` AttributeError: 'int' object has no attribute 'ua_types'` 
Do I need to convert it to bin()? By the error message, does it shouldn't convert already?

zerox1212, 2019-04-10:
I recommend you take some Python Basics online course. The error is very clear that you are calling a method on a simple integer. You must of overwritten the `node` variable, because it is no longer a UA Node object.

lcflorindo, 2019-04-10:
> I recommend you take some Python Basics online course. It's hard to help you without seeing the entire code snippet. Are you sure the `node` variable is really a UA Node object?

It is a UA Node. What the errors means it's more about what should I write to the lib understand it as value than my Py skills.

From a skilled one, what seems wrong at the code below?:
` import opcua
  from opcua import Client, ua

  if __name__ == "__main__": 
      client = Client("opc.tcp://10.12.96.148:49320")
      try:
          client.connect()
          print("Connected to Server")   
          root = client.get_root_node()
          print("Objects node is: ", root)

          node = client.get_node("ns=2;s=L8RC_TMEC_WS00VS_CAMERA.WS.peca_NOK")
          node.set_attribute(ua.AttributeIds.Value, 1)

      finally:
          client.disconnect()`

oroulet, 2019-04-10:
in fact the error in that case is that set_attribute takes a ua.DataValue object as argument, not simply an interger. use Node.set_value() if you want something simple.

lcflorindo, 2019-04-10:
> in fact the error in that case is that set_attribute takes a ua.DataValue object as argument, not simply an interger. use Node.set_value() if you want something simple.

The solution from [https://github.com/FreeOpcUa/python-opcua/issues/708] apparently worked.

closed 2019-04-10 by lcflorindo