FreeOpcUa/python-opcua #822: Request to add a new node

kozachynskyi, 2019-04-17:
Hi Developers, thank you again for you immense work. I explored an interesting issue, that is buried in the depth of the code, that I cannot comprehend.

**Description**: I have a new namespace. In this namespace I have a custom object type that has an object inside of it. 

[image]

**Task**: create a new object in a new namespace based on my custom object type. When I do it from the server side then everything goes as planed: all new nodes are created in my custom namespace.
When I do same thing from client, a nested objected drops to namespace=0. 

[image]

I added a code and a debug information, so you can repeat my experiment. 
[app.log](https://github.com/FreeOpcUa/python-opcua/files/3090542/app.log)

**MyIdeas**: I think the conversion from variant_to_binary somehow erases information about a namespace of an object to create. I also think that this issue is relevant only for objects (not sure though). Why I think so:

When ua_client creates a request, a data that is being packed to binary is correct: `AddNodesRequest.Parameters:AddNodesParameters.RequestedNewNodeId:TwoByteNodeId(ns=2;i=0) `

however when server unpacks a binary, the only difference is that a new object is missing namespace information:

`AddNodesRequest.Parameters:AddNodesParameters.RequestedNewNodeId:TwoByteNodeId(i=0)` 

I followed into the hole until opcua.ua.ua_binary.from_binary, but here I'm useless. Thank you for helping and looking into it.

```
import logging
import time
from opcua import Client, Server
from opcua import Node
from opcua.ua import NumericNodeId, QualifiedName, StringNodeId

def main():
    server = Server()

    try:
        server.start()
        obj_type = server.nodes.base_object_type
        obj_node = server.get_objects_node()
        NS = server.register_namespace('test_namespace')

        TestObjectType = obj_type.add_object_type(StringNodeId('TestObjectType', NS),
                                                  QualifiedName('TestObjectType', NS))
        obj_type.add_object_type(StringNodeId('NestedTestObjectType', NS),
                                 QualifiedName('NestedObjectType', NS))
        TestObjectType.add_object(NumericNodeId(0, NS), QualifiedName('Nested', NS).to_string(),
                                  StringNodeId('NestedTestObjectType', NS)).set_modelling_rule(True)

        # Creating an object
        server_origin = obj_node.add_object(NumericNodeId(0, NS), QualifiedName('CreatedByServer', NS).to_string(),
                                            StringNodeId('TestObjectType', NS))  # type: Node
        instantiated_from_server = server_origin.get_child(QualifiedName('Nested', NS).to_string())
        print (instantiated_from_server)

        client = Client('opc.tcp://admin@localhost:4840')
        client.connect()

        client_origin = client.get_objects_node().add_object(NumericNodeId(0, NS),
                                                             QualifiedName('CreatedByClient', NS).to_string(),
                                                             StringNodeId('TestObjectType', NS))  # type: Node
        instantiated_from_client = client_origin.get_child(QualifiedName('Nested', NS).to_string())
        print(instantiated_from_client)

        while True:
            time.sleep(0.5)

    finally:
        client.disconnect()
        server.stop()

if __name__ == "__main__":
    logging.basicConfig(level=logging.INFO)
    # logging.basicConfig(level=logging.DEBUG, filename='app.log', filemode='w')
    Logger = logging.getLogger(__name__)

    main()
```

kozachynskyi, 2020-03-20:
@oroulet @brubbel Hi, I have a small example to check the issue I posted about. Could you please run it and check if my request makes sense? It shouldn't take more than 5 Minutes, but it can show some hidden bug. Thank you!

brubbel, 2020-03-23:
Fixed in https://github.com/brubbel/python-opcua/tree/fix-instantiate

kozachynskyi, 2020-03-24:
Incredible work! 5 Lines of code, appreciate it, you're the best!

closed 2020-03-24 by kozachynskyi