FreeOpcUa/python-opcua #823: Connection loss by client

OPCle, 2019-04-18:
Hi all, I am trying to identify a problem, which only occurs when using the server of my customer. Unfortunately, I can not  reconstruct this issue, because I have no access to their network/server - which makes the troubleshooting difficult... When using my server, the below error message never occurs. Please notice that the error occurs after my client successfully connected and also read values. 

`WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
ERROR:concurrent.futures:exception calling callback for <Future at 0x4470c18 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\python37\lib\site-packages\opcua\client\ua_client.py", line 463, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\python37\lib\site-packages\opcua\client\ua_client.py", line 88, in check_answer
    hdr.ServiceResult.check()
  File "C:\python37\lib\site-packages\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)
WARNING:opcua.client.ua_client.Socket:ServiceFault from server received while waiting for publish response
ERROR:concurrent.futures:exception calling callback for <Future at 0x4470048 state=finished returned Buffer>
Traceback (most recent call last):
  File "C:\python37\lib\concurrent\futures\_base.py", line 324, in _invoke_callbacks
    callback(self)
  File "C:\python37\lib\site-packages\opcua\client\ua_client.py", line 463, in _call_publish_callback
    self._uasocket.check_answer(data, "while waiting for publish response")
  File "C:\python37\lib\site-packages\opcua\client\ua_client.py", line 88, in check_answer
    hdr.ServiceResult.check()
  File "C:\python37\lib\site-packages\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSessionClosed: The session was closed by the client.(BadSessionClosed)`

The Code I am using looks like this:

`import sys
from threading import Thread
import logging
import time
from opcua import Client, ua
from opcua import Subscription
import xlrd
import os
from os.path import dirname

sys.path.insert(0, "..")

opc_connected = False

# Data change handler for measurement change
class SubHandler1(object):

    def datachange_notification(self, node, val, data):
        print("Data changed.", node, val)
        CallViews()

# Data change handler for server state change
class SubHandler2(object):

    def datachange_notification(self, node, val, data):
        print("OPC UA server state changed.")

# Thread for handling data change
class CallViews(Thread):

    def __init__(self):
        Thread.__init__(self)
        self.daemon = True
        self.running = True
        self.start()

    def stop(self):
        self.running = False

    def run(self):
        while self.running:
[... 231 lines cut ...]
        disconnect()
        print("OPC UA error while trying to read quality.")

if __name__ == "__main__":
    loadSettings()
    # Connect to OPC UA server
    connect()
    read_Laser1_PartID()
    read_Laser1_MaterialID()
    read_Laser1_Positions()

    read_Laser1_Values()
    read_Laser2_Values()
    read_Laser3_Values()
    read_Laser4_Values()

    read_All_Quality()
`
I have some guesses what the problem could be:

1. Creating a new thread called by the subscription is the problem.
2. The subscription event gets called 2x times after connecting to the server.
3. Parameter settings e.g.     params.RequestedMaxKeepAliveCount = 10000 or  params.MaxNotificationsPerPublish = 10000

Thank you very much for your support!

jkbngl, 2020-01-26:
HI @OPCle I am facing the same issue, found anything that has caused the issue?

ColinISW, 2020-01-27:
Hi, @jkbngl Not really..., but I guess that it was caused by to heavy reading. So, what I did (and now it's working):

1. Try it with only reading one value - does the error still occur?

2. I am currently using these parameters for my client: 

```
params = ua.CreateSubscriptionParameters()
params.RequestedPublishingInterval = 500
params.RequestedLifetimeCount = 3000
params.RequestedMaxKeepAliveCount = 10000
params.MaxNotificationsPerPublish = 2147483647
params.PublishingEnabled = True
params.Priority = 0
```

3. Added some time.sleep(0.1) after reading a value so the server can process everything.

I hope it helps you.

zerox1212, 2020-01-27:
Your issue is not very well formatted and is hard to read. I don't think many people will help you with a post like this.

Anyways you are doing some things I definitely don't understand. Why do you need a thread to read values? Why does reading the values have to get the node every time the function is called? Why not just create a subscription for all your laser/quality values?

OPCle, 2020-01-27:
Yes, you're right..sorry - tried to fix the code highlighting.

1. I am creating threads because the [python example](https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py) says for SubHandlers:

>  Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
   
Without these threads it is not working.

2. The values are not very often read (~every 15 minutes during production) and to make a subscription for each value seems to be a little much. Because of that I am subscribing only to one node and read the others after this one changed.

zerox1212, 2020-01-28:
I see. If this code works when using your own network/server then isn't it more likely a network or infrastructure issue?