FreeOpcUa/python-opcua #824: Force change FindServers call reply

PGajewski, 2019-04-18:
Hi,
I'm running OPC-UA server on Docker container and I want use client from external device (PLC). This device knows only the docker host IP address.
It is possible to set DiscoveryURL and EndpointURL with docker host IP? I've try set different hostname for BinaryServer(0.0.0.0) nad InternalServer(docker host IP), but still FindServers call return endpoint with docker container IP, co my client on another machine cannot establish the new session (DiscoveryURL is set properly).

oroulet, 2019-04-18:
This is a docker issue/question. I have no idea

PGajewski, 2019-04-19:
It's not a docker issue, but conflict between endpoints system and NAT translation.
The problem was inside get_endpoints method in InternalServer. It's change response endpoint URL when it was used by UaProcessor with a non default socket argument. I've modify Server and InternalServer classes - add special flag inside InternalServer for disable changing endpoint (replace hostname by socketname) and inside Server - two different endpoints for internal and binary server. Topic closed.

oroulet, 2019-04-19:
Can you make PR. So we can look at it?

oroulet, 2019-04-19:
I thought we were dumbly returning the same string but maybe someone changed that. There is also a open PR to make that stuff really complicated but I never dared committing it

brubbel, 2019-04-22:
If you mean this one: https://github.com/FreeOpcUa/python-opcua/pull/687
I've been using this modification since around september last year without issues.

@PGajewski You can test if it solves your problem using this branch:
https://github.com/brubbel/python-opcua/tree/testing