FreeOpcUa/python-opcua #825: read_raw_history always returns StatusCode: Good

SJG2019, 2019-04-23:
I am not sure if this is an issue from the package or the server but when I am reading the raw history from a server (HyperHistorian) I always get the staus code good. However, I know for sure that there are data in the history where the status code was bad and I get them back with the value "null", as I also see them when looking at them via an OPC UA Client application, but the Status Code is set to good instead of returning the original status code. 

Is this intetionally, if so is there a way I can get the original status code?

Thanks for the help.

zerox1212, 2019-04-23:
Are you talking about the status code of an individual data value in your raw read?

SJG2019, 2019-04-24:
Yes,the response I get looks like this (result saved as a csv): 
```
76.0;StatusCode(Good);2018-03-02 08:34:53.006
;StatusCode(Good);2018-03-02 08:34:55.381
```
While it should look like this (result from ProSys OPC UA Client):

```
2018-03-02T08:34:53.006;2018-03-02T09:34:53.006;76;0x00000000;GOOD
2018-03-02T08:34:55.381;2018-03-02T09:34:55.381;null;0x80000000;BAD
```

zerox1212, 2019-04-24:
I took a look at the code, but I don't have a test system running at the moment and I can't find the problem without out debugging. This gets into the binary protocol which has a lot of recursion and isn't easy to follow.

My guess is that at some point when the server HistoryReadResponse is being parsed the DataValue is being instantiated without passing the status code parameter. So it just defaults to GOOD.

Is the data you are writting to CSV from a `ua.DataValue` object? I can't remember if history has a special UA Type.

SJG2019, 2019-04-24:
Ok, that might be an explanation. 
From the documentation and the link to the source I gathered that the value I get from the raw_history_read is from this [part](https://python-opcua.readthedocs.io/en/latest/_modules/opcua/common/node.html#Node.read_raw_history)
```
result = self.history_read(details)
return result.HistoryData.DataValues
```
I am not sure if this means it is an own UA Type as I am fairly new to the whole subject.

zerox1212, 2019-04-24:
You could try to set a break point and step into the `history_read`. At some point it will get a HistoryReadResult and parse the binary into UA objects (such as DataValue).

Nebukadneza, 2019-05-06:
I think this is exactly #833 — at least this usecase is where we found the StatusCode print issue.
@SJG2019 — can you print `.value` of your statuscode? I’d guess the value is correct, and only the print is broken ^_^.

SJG2019, 2019-05-06:
@Nebukadneza Thanks, that indeed helped, however we now have a numerical value in the status code, which gives a 0 for a good and some seemingly arbitrary number for bad codes. Is there an easy way to get the printed status here? Is there a reference table that could be used for standard OPC UA status codes or something? 
As said before I am fairly new to the topic :)

Nebukadneza, 2019-05-07:
The seemingly arbitrary numbers are the decimal representation of the hex as found here:
https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/ua/status_codes.py#L5
If you print the statuscode itself, you get the `.name`. The bug I was referring to causes the `.name` not being derived from the `.value` at all times, thus messing up the print. The actual `.value` would be correct.

@helmut-jacob is whipping up a fix for that in #835 right now. If you want, you can test out the code from his PR and see if that solves the problem for you.