FreeOpcUa/python-opcua #827: How to instatiate from AnalogItemType

biancode, 2019-04-26:
Hi,

I have a problem to get the properties like range and units to fill up with values.
How to instantiate from AnalogItemType so that I be able to set all properties, please?

idea is, but doesn't work:

I want to use a Type definition:

```
    analogItemType = server.get_root_node().get_child(["0:Types",
                                                       "0:VariableTypes",
                                                       "0:BaseVariableType",
                                                       "0:BaseDataVariableType",
                                                       "0:DataItemType",
                                                       "0:AnalogItemType"])

    analogInput1 = piTypeDefinition.add_object(idx, "AnalogItem", analogItemType.nodeid)
    analogInput1.set_modelling_rule(True)
    
    # maybe here is some todo

```
and an intance from my new type:

```    
pi = objects.add_object(idx, "PI GPIO", piTypeDefinition)

analogInputInstance2 = device.get_child(["{}:Inputs".format(idx), "{}:Analog".format(idx), "{}:Analog_Input2".format(idx)])

# analogInputInstance1Unit = device.get_child(["{}:Inputs".format(idx), "{}:Analog".format(idx), "{}:Analog_Input2".format(idx), "{}:EngineeringUnits".format(idx)])

# analogInputInstance1Unit.set_value("V")

voltageList.append(analogInputInstance1)
```
Thank you!
Klaus

biancode, 2019-04-26:
https://documentation.unified-automation.com/uasdkhp/1.0.0/html/_doc__opc_ua__data_access__variable_types.html#Doc_OpcUa_AnalogItemType

biancode, 2019-04-26:
Is this a good solution?

```
    analogInput = inputsAnalogFolder.add_object(idx, "Analog", analogItemType.nodeid)
    analogInput.set_modelling_rule(True)
    analogInputProperties = analogInput.get_properties()
    for propItem in analogInputProperties:
        propItem.set_modelling_rule(True)
        propItem.set_attr_bit(ua.AttributeIds.AccessLevel, ua.AccessLevel.CurrentRead)
        propItem.set_attr_bit(ua.AttributeIds.UserAccessLevel, ua.AccessLevel.CurrentRead)
```

biancode, 2019-04-26:
warning is:

> WARNING:opcua.common.copy_node:Instantiate: while copying attributes from node type Node(NumericNodeId(i=2368)), attribute AccessLevel, statuscode is StatusCode(BadAttributeIdInvalid)
> WARNING:opcua.common.copy_node:Instantiate: while copying attributes from node type Node(NumericNodeId(i=2368)), attribute UserAccessLevel, statuscode is StatusCode(BadAttributeIdInvalid)

oroulet, 2019-04-26:
What are you trying to do?
From the code above it look like you are trying to create a new type based on AnalogItemType. In theory you should use add_object_type and not add_object. Although there might be bugs, I never heard of anyone using this. 
Then you can use add_object when you really want to instantiate you object

biancode, 2019-04-26:
The AnalogItem is just an object of my type. It is not the inherit from AnalogItemType . I want to get an instance in my type to have analog items with units etc.

biancode, 2019-04-26:
[image]

oroulet, 2019-04-26:
OK so you need to 
mynodetype = add_object_type(whatever ) 
mynodetype.add_object(whatever, nodeid of AnalogItemType)
Does this lead to an error?
If the standard address space xml has errors, it might be that you get some warning, but then you can just overwrite these attributes like you show in https://github.com/FreeOpcUa/python-opcua/issues/827#issuecomment-486955828

biancode, 2019-04-26:
Is there some example how to work with the 'EURange' i=2368 node?

oroulet, 2019-04-26:
I don't think so . There are many nodes with strange name in address space but they are all standard nodes. they do not do anything special

biancode, 2019-04-26:
@oroulet  Thank you!

biancode, 2019-04-26:
An example of how to fully use AnalogItemType would be great.

zerox1212, 2019-04-26:
There are several examples in the folder showing how to create a custom object type and then later calling `instantiate()` to create a new object from the custom type.

I prefer to use a UA modeling tool to design my types. Then I import them from XML when the server starts. After that only `instantiate()` is needed.

Molinsharian, 2020-04-15:
Here is a little example to create an AnalogInputType object with the defined properties in the OPCUA-Specification:

1. Create an object with the objectType= 'AnalogInputType'
```python
var_ait  = base_object.add_object(idx, "ExampleAnalogItemTypeVariable",2368)
```
2. Create python object for eu_range, instrument_range, eu_information
```python
from opcua import ua
eu_range = ua.Range()
eu_range.Low                    = 0
eu_range.High                   = 10
instrument_range = ua.Range()
instrument_range.Low            = 0
instrument_range.High           = 10
eu_information = ua.EUInformation()
eu_information.NamespaceUri     = "http://www.opcfoundation.org/UA/units/un/cefact"
eu_information.UnitId           = 4342098# ID for Common Code BAR
eu_information.DisplayName.Text = bar
eu_information.Description.Text = bar
```

3. Add the values to the 

```python
for i in var_ait.get_properties():
      b = i.get_browse_name()
      if b.to_string() in "0:InstrumentRange": #instrument_range
          data = instrument_range
      elif b.to_string() in "0:EURange": # eu_range
          data = eu_range
      elif b.to_string() in "0:EngineeringUnits" : # eu_information
          data = eu_information
      elif b.to_string() in "0:Definition" : # eu_information
          data = definition
      elif b.to_string() in "0:ValuePrecision" : # eu_information
          data = value_precision
      else:
          raise ValueError('Problems with AnalogItemType')
      i.set_value(data)
```
2368 is the [ObjectId](https://raw.githubusercontent.com/FreeOpcUa/python-opcua/master/opcua/ua/object_ids.py) of the AnalogInputType

Is not the most elegant way but it works.