FreeOpcUa/python-opcua #828: How to subscribe with just nodeID

dhoney, 2019-04-29:
Is it possible, if given a list of only NodeIDs for a client connection to create a subscription and with just those NodeIDs and have the values update every 1000ms (even if unchanged)?

Looking at some examples and getting my QA server setup this week to start trying this out but wanted to ask about the feasibility of this first since I see some other examples do a bit more than just use the NodeID.

Thanks!

zerox1212, 2019-04-29:
You can look at this to see how to use `get_value()`. https://github.com/FreeOpcUa/python-opcua/issues/538

Subscription is not designed for timed updates. If you want to record things on some time base you could write the "recording" function yourself and just rely on subscription to make sure you always have the latest data.

dhoney, 2019-04-29:
Good point on the "recording" function.

I'll look at #538 and see if I can get a subscription on when data changes for the node IDs.