FreeOpcUa/python-opcua #832: Writing the values to multiple nodes at once?

AccellaGroup, 2019-05-04:
I have implemented a client app where I am updating three tags in a PLC via a KEPWare server. This is done by calling the 'node.set_value' method three times in a row. It works, but I would like to speed up the application. Is there a way to do all three updates at once?

I have done a little bit of digging and there seems to be a UAclient.write() method that accepts a list of nodes. Not sure if that would solve my problem. It would be greatly appreciated if anybody here had a coding samples of the UAclient.write() method and/or an explanation of the parameters. Thanks!

oroulet, 2019-05-04:
Look at code of set_value. You can copy and modify to write servers at once. I think there is also an example doing that... Or a bug report

AccellaGroup, 2019-05-05:
Oroulet,
Not sure I follow. Here's the code for set_value from the library:

    def set_value(self, value, varianttype=None):
        datavalue = None
        if isinstance(value, ua.DataValue):
            datavalue = value
        elif isinstance(value, ua.Variant):
            datavalue = ua.DataValue(value)
            datavalue.SourceTimestamp = datetime.utcnow()
        else:
            datavalue = ua.DataValue(ua.Variant(value, varianttype))
            datavalue.SourceTimestamp = datetime.utcnow()
        self.set_attribute(ua.AttributeIds.Value, datavalue)

It seems to me that this only allows me to update the data for one tag only, unless I'm missing something. Could you please elaborate? Thanks!
In the meantime, I'll look through the example and issues to seem if this has been addressed already.