FreeOpcUa/python-opcua #833: Deserialization of StatusCode incorrect doesn't set name & doc

helmut-jacob, 2019-05-06:
Serializing a StatusCode and deserializing it again does not resolve name & doc. Code to reproduce:

`In [32]: sc = StatusCode(StatusCodes.BadNotConnected)
In [33]: b = struct_to_binary(StatusCode(StatusCodes.BadNotConnected))
In [34]: sc2 = struct_from_binary(StatusCode, BytesIO(b))
In [35]: sc
Out[35]: StatusCode(BadNotConnected)
In [36]: sc2
Out[36]: StatusCode(Good)
In [37]: sc.value
Out[37]: 2156527616
In [38]: sc2.value
Out[38]: 2156527616`

StatusCode.value is correct but it's confusing to get StatusCode(Good) on prints even if the value represents a Bad code.

oroulet, 2019-05-06:
that is very strange... I am sure this has worked... and there should be tests for it. Great if you can have a look

helmut-jacob, 2019-05-06:
I mean .name and .doc are not serialized but only filled in the constructor:
```
...
    ua_types = [("value", "UInt32")]

    def __init__(self, value=0):
        if isinstance(value, str):
            self.name = value
            self.value = getattr(status_codes.StatusCodes, value)
        else:
            self.value = value
            self.name, self.doc = status_codes.get_name_and_doc(value)
...
```

The constructor however is not called for deserializing a StatusCode. Only the "value" attribute will be set.

Not sure what the best way to fix this would be. Any ideas?

zerox1212, 2019-05-06:
Do you mean to say that `status_codes.get_name_and_doc(value)` needs to get moved out side of `init`?

helmut-jacob, 2019-05-17:
@oroulet please close this issue as PR is merged.

closed 2019-05-17 by helmut-jacob