FreeOpcUa/python-opcua #836: Authenticate client using certificates/login

Dubehh, 2019-05-07:
I found similar questions but didn't find a working answer. I'd like to authenticate the client before he's able to read data, is this possible? I tried using certificates but a client is able to use whatever certificate he or she desires (meaning it's not validated). I saw [this ](https://github.com/FreeOpcUa/python-opcua/issues/392)reference but i'm not sure if it's the same as what I am looking for.

**Server.py**
```
server = Server()
await server.init()
await server.load_certificate("cert.der")
await server.load_private_key("key.pem")

server.set_security_IDs(["Basic256Sha256"])
server.set_endpoint('opc.tcp://127.0.0.1:4840/freeopcua/server/')
server.set_security_policy([ua.SecurityPolicyType.Basic256Sha256_Sign])
```

**Client.py**
```
client = Client("opc.tcp://127.0.0.1:4840/freeopcua/server/")
client.set_security_string(
        "Basic256Sha256,"
        "Sign,"
        "different_cert.der,"
        "different_key.pem")

client.connect()
```

I used a **differently** generated certificate/key for the client, yet he's still able to fetch data from the server.

If there's a way to authenticate with just username/password i'd be happy aswell..

mihainstr, 2019-05-07:
We had a similar issue and it was related to the policy incorrect handling, check this issue for a possible solution:  #811

Dubehh, 2019-05-07:
> We had a similar issue and it was related to the policy incorrect handling, check this issue for a possible solution: #811

Unfortunately didn't fix it.

zerox1212, 2019-05-07:
Security is waiting for PR from someone who knows certificate handling etc.

Meanwhile you could follow up with this https://github.com/FreeOpcUa/python-opcua/pull/783 .

brubbel, 2019-05-07:
> I used a **differently** generated certificate/key for the client, yet he's still able to fetch data from the server.

Possibly because security policy "NoSecurity" is still allowed. (in parallel with the certificate you have set up). You could remove this setting from server.py.
https://github.com/FreeOpcUa/python-opcua/blob/71ff089d8cf7122a80ac239148797ccb39e35d66/opcua/server/server.py#L117-L121

zerox1212, 2019-05-07:
@brubbel Shouldn't that policy list get overwritten when `server.set_security_policy([ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt])` is called?

brubbel, 2019-05-07:
My mistake, that policy is only for secure communications (i.e. trusted source), not for authentication.