FreeOpcUa/python-opcua #837: [Server/Question] intercept data change for client connection

mooreaa, 2019-05-07:
I'm trying to understand the best way to utilize the opcua server to expose data from our system, but still keep data synchronize with our local data store separate from the opcua server datastore.

On first try, when our local data store (not opc) is updated, we get a notification about the change, so we use that event to update the server node. This is working.

To handle incoming events from remote opcua clients, we used the datachange event. This triggered a call to our local data store to update the value and this works to a limited extent. However, there are cases where our local system rejects the value (by design), in this case, the opcua server and our local data store are no longer synchronized. We are able to correct this, however, the OPCUA value is changed momentarily until our system sends a notification to fix this.

We would ultimately like to intercept incoming client set value events so that we can handle the data synchronization. 

I saw a somewhat similar case in issue #310 but did not fully understand the recommended solution. 

Looking for some guidance on how best to accomplish this.

Thanks,

zerox1212, 2019-05-08:
What are you confused about in https://github.com/FreeOpcUa/python-opcua/issues/310 ? If you use a separate "request" node your server will not update your "read only" value unless you actually want it updated. How I understand your problem is that you are trying to capture all this in a single node. And that single node will get updated when you don't want it updated, thus triggering subscriber events with bad data.

joncutting, 2020-07-28:
Hi, I am jumping into this thread as it exactly addresses the issue I am facing. I spent the last few hours trying to understand what is the recommended way to handle the situation described in the diagram at the top of the thread and I don't think I got any smarter. Is there any way for code which is communicating with the device and updating values in the server to avoid triggering a callback?

zerox1212, 2020-07-29:
My recommendation is to have your client call a method when attempting to set the value. Your local system can test whatever value was supplied by the client and reject it immediately with what the method returns.

joncutting, 2020-07-29:
So I agree that the recommendation from zerox1212 will work if I have control over the client...but in my case the client is not software that I control, it is commercial software and I have to live with the fact that it is reading and writing from the same node. Any other suggestions?

zerox1212, 2020-07-30:
If you can't control the client then you will have to put in some changes deep in this library to intercept the change before it reaches the address space like what was mentioned above.

All other workarounds will require changing how the client requests the change.

oroulet, 2020-07-30:
I think we have two solutions:
* Patch write call to intercept write requests (easy in python)
* Use/extend the service_callback stuff (maybe only in opcua-asyncio)

zerox1212, 2020-07-31:
@oroulet I was thinking about this a little more. Isn't there a way to put limits on a variable/scalar? That way if a client tried to write a value outside the limit it would return Bad Value or some other respone?

swamper123, 2020-08-03:
@zerox1212 There is an Alarm for that called [LimitAlarmType](https://reference.opcfoundation.org/v104/Core/docs/Part9/5.8.11/). But we are not providing such a thing at the moment.