FreeOpcUa/python-opcua #839: How to set values of individual elements inside array

EOFFJM, 2019-05-15:
Hi

I would like to know how I can set values of elements in a node that is an array. 
I defined the node on the server side, made it a 3 dimension array, and made it writable on the client side. 

from opcua import Server
server = Server()
server_url = "opc.tcp://131.246.76.23:4000"
server.set_endpoint(server_url)
name = "OPCUA_SIMULATION_SERVER"
idx = server.register_namespace(name)
objects = server.get_objects_node()
Param = objects.add_object(idx, "Parameters")
mode = Param.add_variable(idx, "mode", [0,0,0])
mode.set_writable()

On the client side I tried to change the value of the second element to 2 using the following code and got a TypeError: 'Node' object does not support indexing error.
When I try to input values for all 3 elements though (eg. mode.set_value([1,2,3]), I don't get any error.

from opcua import Client
url = "opc.tcp://131.246.76.23:4000"
client= Client(url)
client.connect()
print("Client Connected")
mode = client.get_node("ns=2; i=10")
mode[1].set_value(2)

How can I change individual values on the array?
Thanks.

zerox1212, 2019-05-15:
I would have to look into this, but I do not think the "short cut" of `set_value()` can handle this. You probably need to create your own `DataValue` object to set individual array members.

EOFFJM, 2019-05-16:
Thanks for the advice. I don't really know how to use DataValue, but I will try.

oroulet, 2019-05-16:
That is harder than that. You need to construct yourselves a read request structure.... Look at code of set_attribute

EOFFJM, 2019-06-25:
> That is harder than that. You need to construct yourselves a read request structure.... Look at code of set_attribute

Do you know how to do it?

biancode, 2019-06-30:
```
    localeIdArray_node = server.get_node(
        ua.NodeId(ua.ObjectIds.Server_ServerCapabilities_LocaleIdArray))
    localeIdArray_node.set_value(["de-DE", "fr-FR", "es-ES"])

    localIdArrayValues = localeIdArray_node.get_value()
    # work on array 
    localeIdArray_node.set_value(localIdArrayValues)
```

Hi, for me works this very well in the server ...

Happy coding!

EOFFJM, 2019-07-11:
Hello Mr. Landsdorf

Thank you.

Regards
Fuminori Takahashi

closed 2019-08-05 by EOFFJM