FreeOpcUa/python-opcua #840: Question: OPCUA or asyncua?

mariopoeta, 2019-05-17:
Hi,

I am starting a "pet/side" project using opcua in python, but i am confused, the opcua is in maintenance mode but there are releases happening every month while in asyncua last release was in February.

So which one do you suggest to use?  I am going to use python 3.7.

Best Regards,

zerox1212, 2019-05-17:
Do you need to use client or server?

If you are using the server it's already async so use the main lib. If you need the client then you might need to investigate.

Someone else will chime in, but in general this repo is fairly mature and won't have breaking changes in the releases.

The async repo doesn't have a lot of developers.

mariopoeta, 2019-05-17:
I will use client to connect to kepware server.

Thank you for your feedback!

allnode-github, 2019-08-12:
> I will use client to connect to kepware server.
> 
> Thank you for your feedback!

Hello, I also use asyncua as a client to connect to the kepware opc server, but after copying the program always error, I do not know why?
If you use opcua, there is no problem and you can use it normally.
Do you help me or write a segment asyncua as a client to connect to kepware routines?

allnode-github, 2019-08-12:
d:/web/opcua20190811.py:30: RuntimeWarning: coroutine 'Client.connect' was never awaited
  client.connect()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Root is Node(TwoByteNodeId(i=84))
childs of root are:  <coroutine object Node.get_children at 0x000001DD459A34C8>
d:/web/opcua20190811.py:33: RuntimeWarning: coroutine 'Node.get_children' was never awaited
  print("childs of root are: ", root.get_children())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
name of root is <coroutine object Node.get_browse_name at 0x000001DD459A34C8>
d:/web/opcua20190811.py:34: RuntimeWarning: coroutine 'Node.get_browse_name' was never awaited
  print("name of root is", root.get_browse_name())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
childs og objects are:  <coroutine object Node.get_children at 0x000001DD459A34C8>
d:/web/opcua20190811.py:36: RuntimeWarning: coroutine 'Node.get_children' was never awaited
  print("childs og objects are: ", objects.get_children())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback

Jayant-Pancholi, 2020-06-07:
Hi ,
I am trying to write a server code in python using Asyncua.

I am facing below error.
INFO:asyncua.server.internal_session:Created internal session Internal
INFO:asyncua.server.internal_session:Created internal session Internal
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15958), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15959), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15960), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15961), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15962), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15963), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15964), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=16134), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=16135), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=16136), requested parent node NumericNodeId(i=15957) does not exists
Using IP Address to launch Server:172.23.23.199
IIIIDDDDDDDDDDDXXXXXXXXXX 2
INFO:asyncua.common.manage_nodes:create_method Node(NumericNodeId(ns=2;i=12))
INFO:asyncua.common.manage_nodes:create_method Node(NumericNodeId(ns=2;i=12))
&&&&&&&&&&&&&&&&& Node(NumericNodeId(ns=2;i=31))
INFO:asyncua.common.manage_nodes:create_method Node(NumericNodeId(ns=2;i=31))
INFO:asyncua.common.manage_nodes:create_method Node(NumericNodeId(ns=2;i=31))
##################################################################
@@@@@@@@@@@@@@ Node(NumericNodeId(ns=2;i=39))
Traceback (most recent call last):
  File "ServerExample.py", line 333, in <module>
    loop.run_until_complete(ONE())
  File "C:\Python37\lib\asyncio\base_events.py", line 579, in run_until_complete
    return future.result()
  File "ServerExample.py", line 257, in ONE
    myevgen = await JAY.get_event_generator(Pump2, eventType)
  File "C:\Python37\lib\site-packages\asyncua\server\server.py", line 450, in get_event_generator
    await ev_gen.init(etype, emitting_node=emitting_node)
  File "C:\Python37\lib\site-packages\asyncua\server\event_generator.py", line 43, in init
    self.event = await events.get_event_obj_from_type_node(node)
  File "C:\Python37\lib\site-packages\asyncua\common\events.py", line 190, in get_event_obj_from_type_node
    parent_identifier, parent_eventtype = await _find_parent_eventtype(node)
  File "C:\Python37\lib\site-packages\asyncua\common\events.py", line 224, in _find_parent_eventtype
    raise UaError("Parent of event type could notbe found")
asyncua.ua.uaerrors._base.UaError: Parent of event type could notbe found

Please can anyone tell me why this continuous error is coming and why does it says - Parent node does not exist.

AndreasHeine, 2020-06-07:
for errors which say something was never awaited means you have to put simply a await befor calling coro

AndreasHeine, 2020-06-07:
@Jayant-Pancholi do you use custom nodesets? looks like in order to setup namespace the parent node is not existing!

Jayant-Pancholi, 2020-06-07:
@Andreas, I am following simple skeleton code steps to setup namespace.
How am I suppose to set up parent node from my end.

AndreasHeine, 2020-06-07:
hard to guess without knowing your code ... could be something with asyncio while setup namespace typically the namespace has to setup  node by node or with import xml nodeset if in case of parent node does not exist could be also that your coro for parent node hands over to the eventloop before finishs adding parent node in this case the parent node would not be existing when eventloop goes to the next node!?

AndreasHeine, 2020-06-07:
NumericNodeId(i=11715) is part of "Opc.Ua.NodeSet2.xml" so you have to import the xml file in order to use it!

swamper123, 2020-06-08:
@Jayant-Pancholi I don't know how strict stuff is here, but I would expect async server errors being reported in the opcua-asyncio repo, not here.