FreeOpcUa/python-opcua #842: Using pub sub for faster communication

EOFFJM, 2019-05-19:
Right now I have a opc server and some opc clients that communicate with each other using the set_value and get_value functions. I would like to know if I can use publish/subscribe functions (e.g. create_subscription) to make that communication faster.\
Thanks.

zerox1212, 2019-05-19:
Subscriptions are much faster. You should always use subscription for items that change often.

EOFFJM, 2019-05-20:
Thanks.  I used subscriptions to get values from the server to client.
Right now I also use the set_value function on the clients to change values of some nodes. Is there a way to speed that up as well?

EOFFJM, 2019-05-20:
```
from opcua import Client
from opcua import ua
import time

#Client initial settings
url = "opc.tcp://131.246.76.23:4000"
client= Client(url)
client.connect()
print("Client Connected")

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        print(val)

    def event_notification(self, event):
        print("Python: New event", event)

myvar = client.get_node("ns=2;i=33")
handler = SubHandler()
sub = client.create_subscription(0, handler)
handle = sub.subscribe_data_change(myvar)

```

Hi. This is the code that I have on the client. It is used to get values of the nodes to the clients.
When I run it I get values on the Python Shell but I want to be able to use the values taken from the server to be used in the code. Does anyone know how I can do that? IFor example, in the attachement 7 is the value of node with node id "ns=2;i=33". I want to use this value in my code.
[image]

zerox1212, 2019-05-21:
In `def datachange_notification(self, node, val, data):` save the value to a variable. It isn't very convenient to record the data however so you have to come up with a solution. A quick and dirty way would be to save `val` to a `global` variable.

EOFFJM, 2019-05-21:
```
from opcua import Client
from opcua import ua
import time

#Client initial settings
url = "opc.tcp://131.246.76.23:4000"
client= Client(url)
client.connect()
print("Client Connected")

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        global gval
        print(val)
        gval = val

    def event_notification(self, event):
        print("Python: New event", event)

myvar = client.get_node("ns=2;i=33")
handler = SubHandler()
sub = client.create_subscription(0, handler)
handle = sub.subscribe_data_change(myvar)
print (gval)
```
I added  

global gval
gval = val

in datachange_notification(self, node, val, data):

and 

print (gval) 

on the last line of the code but I get this.
[image]

How do I fix this?
I want to pass val to global variable gval and print it in my code.
Thanks.

einarhh, 2019-05-21:
You need to define the variable in the correct scope.
Add gval = None above myvar.

See: https://pythonspot.com/scope/

EOFFJM, 2019-05-21:
Oh. I see. Thanks!

einarhh, 2019-05-21:
But it would probably be better to pass the value to a function from datachange_notification.

EOFFJM, 2019-05-21:
> But it would probably be better to pass the value to a function from datachange_notification.

How do I pass the value to a function from datachange_notification?

EOFFJM, 2019-05-21:
```
from opcua import Client
from opcua import ua
import time

#Client initial settings
url = "opc.tcp://131.246.76.23:4000"
client= Client(url)
client.connect()
print("Client Connected")

class SubHandler(object):
    
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        global gval
        #print(val)
        gval = val
        
class SubHandler2(object):
    
    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another 
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        global gval2
        #print(val)
        gval2 = val
    
    
gval = None
gval2 = None
myvar = client.get_node("ns=2;i=6")
myvar2 = client.get_node("ns=2;i=8")
handler = SubHandler()
handler2 = SubHandler2()
sub = client.create_subscription(0, handler)
sub2 = client.create_subscription(0, handler2)
handle = sub.subscribe_data_change(myvar)
handle2 = sub2.subscribe_data_change(myvar2)
while True:
    print (gval,gval2)
    time.sleep(0.1)
```

Hi. I was able to take values from one node using subscription. Now I want to take values from two nodes. I created this code to do just that and it works, but I think it looks pretty bad. Does anyone have any suggestions to make it better? I attached the shell results for this code. I am thinking of using up to about 8 nodes in the future.
Thanks.
[image]

einarhh, 2019-05-21:
You should really study up on python.
There is a lot of ways you could do it, depending on your needs.

Here is two possible ways, using value in a function and saving values to a dictionary.

`from opcua import Client
from opcua import ua
import time

#Client initial settings
url = "opc.tcp://131.246.76.23:4000"
client= Client(url)
client.connect()
print("Client Connected")

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    data_change and event methods are called directly from receiving thread.
    Do not do expensive, slow or network operation there. Create another
    thread if you need to do such a thing
    """

    def datachange_notification(self, node, val, data):
        # Send the value to a function
        use_value(val,node)
        # Store the value in a dict
        dict[node] = val

# Using the value in a function
def use_value(value,node):
    print(str(node)+" has a new value: "+str(value))

# New empty dict
dict = {}

myvar = client.get_node("ns=2;i=6")
myvar2 = client.get_node("ns=2;i=8")

handler = SubHandler()
sub = client.create_subscription(0, handler)

handle1 = sub.subscribe_data_change(myvar)
handle2 = sub.subscribe_data_change(myvar2)

while True:
    # Loop throug the dict and print values
    for item in dict:
        print(str(item)+" has value "+str(dict[item]))

    time.sleep(0.1)
`

EOFFJM, 2019-05-21:
Thank you. Can someone please explain to me where the method datachange_notification(self, node, val, data): is being called? I am curious because I cannot find it anywhere.

zerox1212, 2019-06-03:
I don't remember exactly, but it's called from the network thread and you shouldn't need to worry about where it's coming from.