FreeOpcUa/python-opcua #847: Read data from OPC-UA server as soon it gets updated on UA server in python

amitChechare, 2019-06-03:
I have written OPC-UA server in python, In that server receives data from meters. Now I am writing OPC-UA client in which I want to read data for meter  which gets sampled. i.e I want to read data from OPC-UA server as soon as it gets that data from meter. Can any one give me idea how to do that.

oroulet, 2019-06-03:
well this is what subscription is about, isnt'it?

EOFFJM, 2019-06-03:
[https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py](url)
Try this example code for subscription.

amitChechare, 2019-06-04:
> [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py](url)
> Try this example code for subscription.

this link blank.

amitChechare, 2019-06-04:
> well this is what subscription is about, isnt'it?

Thank u for that........

amitChechare, 2019-06-04:
> [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py](url)
> Try this example code for subscription.

can you please share the link of complete example of OPC UA Client.  As I am new in both OPC and python. It will help me a lot.

EOFFJM, 2019-06-04:
> > [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py](url)
> > Try this example code for subscription.
> 
> this link blank.

https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py

amitChechare, 2019-06-05:
> > > [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py](url)
> > > Try this example code for subscription.
> > 
> > 
> > this link blank.
> 
> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py

Thank you for your link that you gave me, It gave me some idea about my issue. But I have below query in code given in above link. 

#Link code line no 68. FYR.
uri = "OPCUA_SERVER1" # Name of name space use in my UA server
idx = client.get_namespace_index(uri)

I am not getting any idea that how should I create  " ** myvar** " and " **obj** " in my case using my namespace i.e "OPCUA_SERVER1". I got three namespace 1) http://opcfoundation.org/UA/ 2)urn:freeopcua:python:server 3) OPCUA_SERVER1 when I printed that **client.get_namespace_array()**
Name space which I am using is correct...?

#myvar = root.get_child(["0:Objects", "{}:MyObject".format(idx), "{}:MyVariable".format(idx)])
#obj = root.get_child(["0:Objects", "{}:MyObject".format(idx)]) 

Thanks in advance.

KrahmerTUD, 2019-08-20:
I guess you should use the namespace, which you register at your server before, i.e. "OPCUA_SERVER1" in your case.