FreeOpcUa/python-opcua #848: Standard namespace - Invalid attributes for HasSubType reference

1000stSheep, 2019-06-04:
I am just a pythonist trying to use OPC UA. First big thanks for your work.

According to the UA standard [0], the HasSubType reference should neither be abstract nor symmetric [1].
However the NodeSet used for codegeneration does not explicitly define attribute values [2], so generated python code does not explicitly set the attributes. [3]
BUT: Server publishes both attributes 'Symmetric and 'IsAbstract' as true. [4]

So either NodeSet is wrong at that specific (missing attribute values) type, or python-opcua uses wrong defaults for (at least abstract and symmetric) attributes.

Former could be (was at least once, see [1]).
The latter would affect a lot of other places and usage and i cannot overlook consequences of changing that.

[0] published PDFs only accessible for registered companies
[1] https://forum.unified-automation.com/topic1631.html
[2] https://github.com/FreeOpcUa/python-opcua/blob/master/schemas/Opc.Ua.NodeSet2.Part3.xml#L374
[3] https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/standard_address_space/standard_address_space_part3.py#L911
[4] Attached Screenshot from UaExpert
[image]

oroulet, 2019-06-04:
Yes they should be false. Fix welcome. Probably somewhere in xml import

zerox1212, 2019-06-04:
Probably should be false as stated. Maybe was a result of one of several contradictions in the OPC UA spec.

1000stSheep, 2019-06-05:
So possible fix would be changing defaults for attributes IsAbstract and Symmetric in python class ReferenceTypeAttributes to False:
https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/ua/uaprotocol_auto.py#L5451

I modified my pip installed module, checked a few reference types against UA Demo server, looking good so far. Could please someone check, too?

No git knowledge or experience on my side, so unable to provide PR or patch, sorry.
Could please someone commit that change? Thanks.