FreeOpcUa/python-opcua #850: SourceTimestamp

shamalb, 2019-06-05:
Hello.
There is found that SourceTimestamp and ServerTimestamp of published tag`s value on instance freeopcua don`t update in case changed value from an external opc-client. 
Tested it with UaExpert.
Explane me, please. There is required any tunings or upgrade freeopcua maybe?

shamalb, 2019-06-07:
[image]
Thanks in advance

oroulet, 2019-06-07:
As far as I remember we donnot write this attributes. but if you write it explicitly for example in python:

dv = ua.DataValue(2)
dv.SourceTimestamp = datetime.now()
myvar.set_value(dv)

then you will see it updated from uaexpert or any other client

shamalb, 2019-06-10:
yes, I agree with you, but it working when all variables placed in server side.
we need to use sourcetimestamp in our some tasks. we take some arguments with timestamp from client side and it very important for us.

oroulet, 2019-06-10:
On server side you should use server.set_attribute_value and you can set source or ServerTimestsamp the exact same way

shamalb, 2019-06-10:
Maybe I did not say correctly.
But can server set sourcetimestamp or servertimestamp equal a local time in case server take changed value of tag from client side, but client did not set timestamp of tag?
For example: if client set value of tag without timestamp then server should setup timestamp before update published tag for other clients

oroulet, 2019-06-10:
>> if client set value of tag without timestamp then server should setup timestamp before update published tag for other clients

Is that part of spec? I did not check but I was expecting that if clients want to write something without timestamsp they should be able to do so

shamalb, 2019-06-10:
Look, in spec "OPC Unified Architecture, part 4" - chapter 7.7 
DataValue is defined ServerTimestamp: "In the case where the OPC UA Server subscribes to a value from another OPC UA Server, each Server applies its own serverTimestamp..."
But Client can write value too if it allowed on server side -  is a same subscribe in our case?

shamalb, 2019-06-11:
I am bookworm maybe, but the question is open

zerox1212, 2019-06-11:
Hard to understand the spec here. It could mean what you say, that server _always_ updates server timestamp, or it could mean that the timestamp is only updated when multiple servers are subscribed to each other (and has nothing to do with the client)

oroulet, 2019-06-11:
It makes sense to write server timestamp when a client write things. Maybe we should always overwrite it... Not sure

shamalb, 2019-06-13:
image it.  You are owner of newspapers and somebody write an article to you. You update timestamp on each issue of newspaper.
If we did not control the timestamp of a write how will we know that is actually now. A write is so old maybe.

shamalb, 2019-06-13:
so are you ready help us?

oroulet, 2019-06-13:
PR is welcome, I propose that you add a check for ServerTimestamp jsut befoire that line and add it if missing. Do NOT write SourceTimestamp, this is the client or subsystem responsability

shamalb, 2019-06-13:
We did it for ServerTimestamp. It worked.
Thank You

closed 2019-06-13 by shamalb

oroulet, 2019-06-13:
Did you open a pull request?

mbay-SAG, 2020-12-10:
We came across that bug as well. If you use an OPCUA Client that only writes the value we changed the write function in address_space.py to:

`def write(self, params, user=UserManager.User.Admin):
        self.logger.debug("write %s as user %s", params, user)
        res = []
        for writevalue in params.NodesToWrite:
            if user != UserManager.User.Admin:
                if writevalue.AttributeId != ua.AttributeIds.Value:
                    res.append(ua.StatusCode(ua.StatusCodes.BadUserAccessDenied))
                    continue
                al = self._aspace.get_attribute_value(writevalue.NodeId, ua.AttributeIds.AccessLevel)
                ual = self._aspace.get_attribute_value(writevalue.NodeId, ua.AttributeIds.UserAccessLevel)
                if not ua.ua_binary.test_bit(al.Value.Value, ua.AccessLevel.CurrentWrite) or not ua.ua_binary.test_bit(ual.Value.Value, ua.AccessLevel.CurrentWrite):
                    res.append(ua.StatusCode(ua.StatusCodes.BadUserAccessDenied))
                    continue
            tz = pytz.timezone('Europe/Berlin')
            writevalue.Value.ServerTimestamp = datetime.now(tz)
            #writevalue.Value.SourceTimestamp = datetime.now(tz)
            res.append(self._aspace.set_attribute_value(writevalue.NodeId, writevalue.AttributeId, writevalue.Value))
        return res`

Make sure to import pytz .
In this scenario the OPCUA Server sets the Servertimetimestamp, which is from our perspective as specified by the OPCUA guidline. Please consider implementing this.

If the Sourcetimestamp is not set by the client, you can use the commented line with SourceTimestamp as well. Hope this helps out in the field.