FreeOpcUa/python-opcua #852: Why no compression in the opc-ua binary transport protocol?

brubbel, 2019-06-24:
Does anyone have an idea why (streaming) compression is not part of the OPC-UA standard? At least I could not find any references.
It could be for security reasons (cfr. BREACH) but I highly doubt this.

zerox1212, 2019-06-25:
Maybe because UA is supposed to be able to scale all the way down to tiny SoC hardware that wouldn't be able to do compression?

brubbel, 2019-06-25:
Yes, that is a valid remark, but it could be made optional.
For example, there could be ''CompressedTcpTransport'' flag in the discovery capabilities.

I did some testing with zstd streaming and large histories, and for small DataValues there is a compression ratio up to 5, which speeds up history transfers considerably.

zerox1212, 2019-06-26:
I like the idea. I have never heard of zstd but it looks interesting and faster than what I'm familiar with from web technologies (gzip).

I suppose you would need to write in to OPC Foundation and wait 2 years for the next standard lol.