FreeOpcUa/python-opcua #856: Problem Loading Type definitions

startpyy, 2019-07-03:
Hi there,

I am having trouble with loading in the Type Definitions from a Server.
The OPC-UA Server was not setup by myself.  Now I am trying to connect to the Server and  read in all the Varaibles and Messages with a Client(Basic256). With .get_value() I get:

`[ExtensionObject(TypeId:FourByteNodeId(ns=2;i=5003), Encoding:1, 76 bytes)]`

So my Client need to know the Extension Object. The Problem is I can't load the type definitions. After connecting and running client.load_type_defitnitions() I always get:

```
client.load_type_definitions()
Traceback (most recent call last):

  File "<ipython-input-2-334a68facbea>", line 1, in <module>
    client.load_type_definitions()

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\client.py", line 565, in load_type_definitions
    return load_type_definitions(self, nodes)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\common\structures.py", line 248, in load_type_definitions
    for desc in server.nodes.opc_binary.get_children_descriptions():

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\common\node.py", line 337, in get_children_descriptions
    return self.get_references(refs, ua.BrowseDirection.Forward, nodeclassmask, includesubtypes)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\common\node.py", line 366, in get_references
    results = self.server.browse(params)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 306, in browse
    data = self._uasocket.send_request(request)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 76, in send_request
    future = self._send_request(request, callback, timeout, message_type)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\client\ua_client.py", line 66, in _send_request
    msg = self._connection.message_to_binary(binreq, message_type=message_type, request_id=self._request_id)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\common\connection.py", line 214, in message_to_binary
    token_id=token_id)

  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\common\connection.py", line 93, in message_to_chunks
    if message_type == ua.MessageType.SecureOpen:

  File "C:\ProgramData\Anaconda3\lib\enum.py", line 326, in __getattr__
    raise AttributeError(name) from None

AttributeError: SecureOpen
```

I tired client.open_secure_channel(), which gave my :
`NotEnoughData: Not enough data left in buffer, request for 4, we have Buffer(size:0, data:b'')`

With client.open_secure_channel(renew=True) it this error vanished. 
I also know how the structure is set up:

ID - String
Time - utcTime
Aktiv - bool
Info - LocalizedText
Text - LocalizedText

Does anybody have an Idee to fix this?
Thank you

zerox1212, 2019-07-03:
Is there any chance you can test this with a regular Python install with only necessary opcua packages installed instead of Anaconda3?

startpyy, 2019-07-04:
I tired it on my raspberry Pi. The File ist basically:

```
import logging
import time

from opcua import Client

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    client = Client("opc.tcp://192.168.1.x:4840/")
    client.set_security_string("Basic256,SignAndEncrypt,OpcUaClient[123].der,OpcUaClient[123].pem")
    client.load_client_certificate("OpcUaClient[123].der")
    client.load_private_key("OpcUaClient[123].pem")
    try:
        client.connect()
        client.load_type_definitions()

    finally:
        client.disconnect()
```
I fixed the IP and certifcates. It easier to read.
The Output is the Same:

```
WARNING:opcua.crypto.security_policies:DEPRECATED! Do not use SecurityPolicyBasic256 anymore!
Traceback (most recent call last):
  File "/home/pi/Schreibtisch/OPCUA/read_opcua_values.py", line 25, in <module>
    client.disconnect()
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/client.py", line 271, in disconnect
    self.close_session()
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/client.py", line 476, in close_session
    return self.uaclient.close_session(True)
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/ua_client.py", line 291, in close_session
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/ua_client.py", line 76, in send_request
    future = self._send_request(request, callback, timeout, message_type)
  File "/usr/local/lib/python3.5/dist-packages/opcua/client/ua_client.py", line 66, in _send_request
    msg = self._connection.message_to_binary(binreq, message_type=message_type, request_id=self._request_id)
  File "/usr/local/lib/python3.5/dist-packages/opcua/common/connection.py", line 214, in message_to_binary
    token_id=token_id)
  File "/usr/local/lib/python3.5/dist-packages/opcua/common/connection.py", line 93, in message_to_chunks
    if message_type == ua.MessageType.SecureOpen:
  File "/usr/lib/python3.5/enum.py", line 274, in __getattr__
    raise AttributeError(name) from None
AttributeError: SecureOpen
```

So is there any connection to the old SecurityPolicy?

startpyy, 2019-07-04:
So I tired some things and suddenly I noticed:

If I am running my Code with load_type_definitions:

```
    client.connect()
    root = client.get_root_node()
    client.load_type_definitions()
    
    var6 = client.get_node("ns=2;s= 29xxxxxxxxxxxxxxxxxs")
#    print(var6.get_value())
#    client.disconnect()
```
The load seems to work. But if i try to print the value, get the value or disconnect always the SecoreOpen statment is raised.
With a new Python-Kernel and without load_type_definitions() I can print, get values or disconnect without any ErrorMessage.

ankitsakhuja7, 2019-08-16:
Hello, did you find the solution how to set the value by client .
Thanks

ArkasDev, 2020-11-03:
@ankitsakhuja7 I am also looking for a solution.