FreeOpcUa/python-opcua #857: Changes to DataTypeDictionaryBuilder break historical extensionobjects

brubbel, 2019-07-04:
https://github.com/FreeOpcUa/python-opcua/commit/aae74aa8c9f77ec2fb67d575c09a1aae81178a2f#diff-f44c14815167ffabcf4efa591593ecd6R236

TLDR; The extensionobject's typeid is binary encoded in history using extensionobject_to_binary, however, above changes to the default nodeid's result in reloading historical extensionobjects build with older versions of freeopcua.

oroulet, 2019-07-17:
What was the default before? As far as I remember that code was using its own automated nodeid generation using some internal counter. That code was a duplication of the existing way to generated nodeid on the fly. None of these methods guarantee the same nodeid if you regenerate structures, any structure change will change the nodeids

brubbel, 2019-07-19:
Before this was also an issue, but it emerged for me only since the changes due to the different "semi-random" ID that was generated.

The only reliable way to use history+custom structures is either to let the user choose a fixed ID or to find a way to store the custom structure also in the database. I tested this but did not find an easy way to reload this custom structure into the server (it is available in the namespace but more actions are needed to make the server aware of + decode these structures in struct_from_binary calls)

For me not a problem, I reverted to the original code, but I think this issue should be tackled in a satisfactory manner.

zerox1212, 2019-07-19:
I don't know extension object very well, but is there no way for the user to supply a "fixed" node ID when the extension object is defined instead of always using the "semi-random" one?

oroulet, 2019-07-19:
@brubbel @zerox1212 yes being able to provide a nodeid sounds like the solution that makes most sense. Can someone make a PR?