FreeOpcUa/python-opcua #859: Only Security Policy Basic256Sha256 is supported for Client. Any way to use other policies??

raainishant, 2019-07-08:
Is there any way to use client with Basic256????
 It Says "Deprecated Do not Use Securitypolicy256 anymore" .
I need to test my OPC Server against all possible Security Policies. 

Thanks

oroulet, 2019-07-08:
I remember someone wanted to remove it. I said we should keep it. He did, but I have never tested it

NewBeePyPy, 2019-07-08:
Thanx @oroulet  for prompt response. I think it has been removed.
I have further questions:
1. Is there any way on Low level clients to still use it? If you can give some direction i can try and if successful we can add as an example or something.
2. Are there any security related threats for using other policies? Why were they removed , if you have any info about that?

zerox1212, 2019-07-08:
I seem to remember that some policies were actually removed from the `crypto` library because they are outdated and insecure.

NewBeePyPy, 2019-07-09:
Thanx @zerox1212.

zerox1212, 2019-07-09:
My mistake.

sb934, 2020-07-20:
I am unable to use either Basic256 or Basic128Rsa15 with my client using freeopcua. Is there any workaround for this issue? The server I am trying to connect to only supports these 2 policies... Did you guys end up removing all support for these? @oroulet

AndreasHeine, 2020-07-20:
they are depricated since opcua spec. v1.04 ...

sb934, 2020-07-20:
So basically no way for me to connect to this server using python? :-1:

AndreasHeine, 2020-07-20:
as far as i saw you still can set the securitystring to a depricated algo. but the deprication came from opc foundation!!!

oroulet, 2020-07-20:
What is the issue? What happens when you try to use these encryptions?

sb934, 2020-07-20:
_Server.py - security string_ 

`server.set_security_policy([
    ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
    ua.SecurityPolicyType.Basic128Rsa15_Sign,
    ua.SecurityPolicyType.Basic256_SignAndEncrypt
])`

The server runs fine, however client throws an error using any of the above mentioned policies...

> opcua.ua.uaerrors._base.UaError: No matching endpoints: 3, http://opcfoundation.org/UA/SecurityPolicy#Basic256

_Client.py_ 
`client.set_security_string("Basic256,SignAndEncrypt,certificate.der,privateKey.pem")`

AndreasHeine, 2020-07-20:
server and client configuration arent the same...

AndreasHeine, 2020-07-20:
shit... sorry

sb934, 2020-07-20:
Haha no worries. I am new to this, working on a school project. Can you help me on how to use an older version of opcua if that solves the problem? I just pip installed freeopcua and started using it...

AndreasHeine, 2020-07-20:
give me a sec. i switht to my laptop setting up a sample :) lets see!

sb934, 2020-07-20:
I can share my simulation server-client code with you if I can get your email?

AndreasHeine, 2020-07-20:
```python
    server.set_security_policy([
                                    ua.SecurityPolicyType.NoSecurity,
                                    ua.SecurityPolicyType.Basic128Rsa15_Sign,
                                    ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
                                    ua.SecurityPolicyType.Basic256_Sign,
                                    ua.SecurityPolicyType.Basic256_SignAndEncrypt,
                                    ua.SecurityPolicyType.Basic256Sha256_Sign,
                                    ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt
                                ])
```

[image]

ok look like its missing... let me figure out what to change!

looks like the client is fine... server doesnt expose the endpoint!?
is the server aslo python-opcua???

AndreasHeine, 2020-07-20:
in server.py you needs to be added

```python

            if ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic128Rsa15,
                                    ua.MessageSecurityMode.SignAndEncrypt)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic128Rsa15,
                                                               ua.MessageSecurityMode.SignAndEncrypt,
                                                               self.certificate,
                                                               self.private_key)
                                     )
            if ua.SecurityPolicyType.Basic128Rsa15_Sign in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic128Rsa15,
                                    ua.MessageSecurityMode.Sign)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic128Rsa15,
                                                               ua.MessageSecurityMode.Sign,
                                                               self.certificate,
                                                               self.private_key)
                                     )
            if ua.SecurityPolicyType.Basic256_SignAndEncrypt in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic256,
                                    ua.MessageSecurityMode.SignAndEncrypt)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic256,
                                                               ua.MessageSecurityMode.SignAndEncrypt,
                                                               self.certificate,
                                                               self.private_key)
                                     )
            if ua.SecurityPolicyType.Basic256_Sign in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic256,
                                    ua.MessageSecurityMode.Sign)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic256,
                                                               ua.MessageSecurityMode.Sign,
                                                               self.certificate,
                                                               self.private_key)
                                     )
```

give me a sec. i put it together end post it so you can replace server.py!

sb934, 2020-07-20:
Yes sir. Everything is coded in python using opcua. See I know everything works fine with Basic256Sha256 or no SecurityPolicy but the instrument I am trying to connect it to is kinda old... I need my client.py to work on either of the other 2 encryption(s)

AndreasHeine, 2020-07-20:
[image]

now it looks better :)

this path sould be kind of similar but with your user and your python-version:
C:\Users\andre\AppData\Local\Programs\Python\Python37\Lib\site-packages\opcua\server\server.py
open and replace all!

```
"""
High level interface to pure python OPC-UA server
"""

import logging
from datetime import timedelta, datetime
try:
    from urllib.parse import urlparse
except ImportError:
    from urlparse import urlparse

from opcua import ua
# from opcua.binary_server import BinaryServer
from opcua.server.binary_server_asyncio import BinaryServer
from opcua.server.internal_server import InternalServer
from opcua.server.event_generator import EventGenerator
from opcua.server.user_manager import UserManager
from opcua.server.discovery_service import LocalDiscoveryService
from opcua.common.node import Node
from opcua.common.subscription import Subscription
from opcua.common.manage_nodes import delete_nodes
from opcua.crypto import security_policies
from opcua.common.event_objects import BaseEvent
from opcua.common.shortcuts import Shortcuts
from opcua.common.structures import load_type_definitions, load_enums
from opcua.common.xmlexporter import XmlExporter
from opcua.common.xmlimporter import XmlImporter
from opcua.common.ua_utils import get_nodes_of_namespace
use_crypto = True
try:
    from opcua.crypto import uacrypto
except ImportError:
    use_crypto = False

class Server(object):

    """
    High level Server class

    This class creates an opcua server with default values

    Create your own namespace and then populate your server address space
    using use the get_root() or get_objects() to get Node objects.
    and get_event_object() to fire events.
    Then start server. See example_server.py
    All methods are threadsafe

    If you need more flexibility you call directly the Ua Service methods
    on the iserver  or iserver.isession object members.

    During startup the standard address space will be constructed, which may be
    time-consuming when running a server on a less powerful device (e.g. a
    Raspberry Pi). In order to improve startup performance, a optional path to a
    cache file can be passed to the server constructor.
    If the parameter is defined, the address space will be loaded from the
    cache file or the file will be created if it does not exist yet.
    As a result the first startup will be even slower due to the cache file
    generation but all further start ups will be significantly faster.

    :ivar product_uri:
    :vartype product_uri: uri
[... 573 lines cut ...]
        Returns:
        """
        self.iserver.isession.add_method_callback(node.nodeid, callback)

    def load_type_definitions(self, nodes=None):
        """
        load custom structures from our server.
        Server side this can be used to create python objects from custom structures
        imported through xml into server
        """
        return load_type_definitions(self, nodes)

    def load_enums(self):
        """
        load UA structures and generate python Enums in ua module for custom enums in server
        """
        return load_enums(self)

    def set_attribute_value(self, nodeid, datavalue, attr=ua.AttributeIds.Value):
        """
        directly write datavalue to the Attribute, bypasing some checks and structure creation
        so it is a little faster
        """
        return self.iserver.set_attribute_value(nodeid, datavalue, attr)
```

sb934, 2020-07-20:
IT WORKED!! Thank you mate... @AndreasHeine

sb934, 2020-07-22:
@AndreasHeine quick question... does the certificate on client and server have to be same since I do not have access to the server certificate for the initial connection and I am supposed to automatically trust the server certificate to establish a connection? I'm working on a Kaiser Raman product.

AndreasHeine, 2020-07-22:
@sb934 quick answer... in python opcua (python opcua client to python opcuaserver) can be the same, doesnt have to be the same but i would recommend you to generate one for each client and server instance! you can generate yours with open ssl (Type x509v3 with SubjectAltname-Extention )

AndreasHeine, 2020-07-22:
for compatibility reasons we accept all ... we maybe change it in the future!

sb934, 2020-07-22:
Alright, I'll try working on this.  Pls don't change nothing until my project is done hahaha

AndreasHeine, 2020-07-22:
if we change it it would be a configuration option on serverside to maintain compatibility ;)

sb934, 2020-07-22:
Alright @AndreasHeine , it throws an error saying user identity token is not valid. Is there some issue with my client certificate?

AndreasHeine, 2020-07-22:
open a new issue and post your stacktrace and i have a look!

sb934, 2020-07-22:
Issue #1109

ronytesler, 2022-05-24:
Can I use `user_name`, `password` and also `security_mode` and/or `security_mode` but without certificate and private key files?

AndreasHeine, 2022-05-24:
> Can I use `user_name`, `password` and also `security_mode` and/or `security_mode` but without certificate and private key files?

its not possible to use encryption without cert and key!

username and pw should be possible!

deuler69, 2023-10-06:
Hi, I stumbled across this post by chance and wanted to know if this library deliberately only allows encryption to be entered with a key and certificate. UAExpert, for example, allows policy and mode on the Security tab without requiring a key and certificate. 
[image]

I want to build a tool that exports all nodes, but is not limeted to only 20 like UAExpert with basic version..

deuler69, 2023-10-10:
> This is not true. There are 2 kinds of certificates.
> 
> * One is the application certificate, UAExpert always creates this on the first launch. This certificate is always need when using in Security Policy.
> * The second is a certificate used for authentification instead of a username/password, this is what you see in the UAExpert Window.

Do you happen to know anything about the subject matter? So if I want to build an application that has the same security mechanisms as UAExpert and want to continue working with the uaasync library and Python, is there an easy way to create such a (self) signed certificate that is also accepted by the OPC server in any case?