FreeOpcUa/python-opcua #863: How to get list of all published nodes?

HubertLaczak, 2019-07-10:
Hi, I'm writing Client app and I don't want to hardcode methods to get specific node with ids. I would like to ask server which nodes (variables) are published, I want to get list of each nodes. 

How to make it? I was reading example from
https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py
but I am not sure which methods are necessary. Can you give me tips?

oroulet, 2019-07-10:
Every server publishes 1000s of nodes. You do not want to access all of them. The protocol is based on knowing what nodes you want. Ideally using domain specific profiles

brubbel, 2019-07-11:
I would just browse the tree upwards starting from the root node.

zerox1212, 2019-07-11:
You would need to browse the tree recursively to achieve this.

HubertLaczak, 2019-07-12:
When I try with:
`
try:
client.connect()
print ("\nConnected\n")
root = client.get_root_node()
print("%s" %root)
print("Children of root are: %s" %root.get_children())    
`

The answer is:
`Node(TwoByteNodeId(i=84))

Children of root are: [Node(TwoByteNodeId(i=87)), Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86))]
`

Am I right, that ids of my nodes are: 87,85,86? At least I use:
`
var = client.get_node(ua.NodeId(85,2))

val = var.get_value()
`
but there is an error: "The node id refers to a node that does not exist in the server address.."

Could you give me an advise or some example? I read 

> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py

HubertLaczak, 2019-07-12:
When I try with:
_client.connect()
print ("\nConnected\n")
root = client.get_root_node()
print("%s" %root)
print("Children of root are: %s" %root.get_children())_

The answer is:
          _Node(TwoByteNodeId(i=84))
          Children of root are: [Node(TwoByteNodeId(i=87)), Node(TwoByteNodeId(i=85)),                     Node(TwoByteNodeId(i=86))]_

Am I right, that ids of my nodes are: 87,85,86? At least I use:
_var = client.get_node(ua.NodeId(85,2))
val = var.get_value()_

but there is an error: "The node id refers to a node that does not exist in the server address.."

Could you give me an advise or some example? I read 

> https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-example.py

brubbel, 2019-07-12:
```
root = client.get_root_node()
print("Objects node is: ", root)

for chNodeId in root.get_children():
    chNode = client.get_node(chNodeId)
    print(chNode)
```

HubertLaczak, 2019-07-12:
I still have problems. I receive:

     Root node is Node(TwoByteNodeId(i=84))
     NodeClass.Object
     3
     NodeClass.Object
     3
     NodeClass.Object
     3

I copied your code:

     try:
        root = client.get_root_node()
        print("Root node is {:s} " .format(str(root)))

        def browse_recursive(node):
        for childId in node.get_children():
               ch = client.get_node(childId)
               print(ch.get_node_class())
               if ch.get_node_class == ua.NodeClass.Object:
                       print("1")
                       browse_recursive(ch)
               elif ch.get_node_class() == ua.NodeClass.Variable:
                         print("2")
                         try:
                                   print("{bn} has value {val}".format(bn=$
                         except ua.uaerror._auto.BadWaitingForInitialDat$
                                   pass
                else:
                          print("3")
        browse_recursive(root)

It looks like NodeClass.Object != ua.NodeClass.Object?
Normally, I use:
 
        var = client.get_node("ns=2;s=Register1")

to get node and later to get the value. But when I add in the code above:
     
     for childId in node.get_children():
               ch = client.get_node(childId)
               print(ch.get_node_class())
               print(ch.get_browse_name()) #I added this line

I receive:

     NodeClass.Object
     QualifiedName(0:Views)
     3
     NodeClass.Object
     QualifiedName(0:Objects)
     3
     NodeClass.Object
     QualifiedName(0:Types)
     3

brubbel, 2019-07-12:
`ch.get_node_class**()** == ua.NodeClass.Object`
There is a typo in your code.

HubertLaczak, 2019-07-17:
Great work, thank you :)

closed 2019-07-17 by HubertLaczak

twodonkeys, 2020-06-02:
mark

mukaraT, 2022-07-04:
I tried with this below function, it is still not able to read the rest of the node ids :(

```
def browse_recursive(node):
    for childId in node.get_children():
        ch = client.get_node(childId)
        print(ch.get_node_class())
        if ch.get_access_level() & (1<< ua.AccessLevel.CurrentRead.mask):
            if ch.get_node_class() == ua.NodeClass.Object:
                browse_recursive(ch)
            elif ch.get_node_class() == ua.NodeClass.Variable:
                try:
                    print("{bn} has value {val}".format(
                        bn=ch.get_browse_name(),
                        val=str(ch.get_value()))
                    )
                except ua.uaerrors._auto.BadWaitingForInitialData:
                    pass
            
```
Any Ideas/Suggestions, where I could read all the node IDs or pass the exceptions and read further variables?

Regards,
Mukara

brubbel, 2022-07-04:
```
...
                try:
                    print("{bn} has value {val}".format(
                        bn=ch.get_browse_name(),
                        val=str(ch.get_value()))
                    )
                except ua.uaerrors.UaStatusCodeError as e:
                    print(e)
                    pass
....
```

SHAFIT, 2023-02-04:
I am trying to browse all the nodes in the Server. It is understood that there are some node which are from the server itself and was able to come up with a recusive function to itrate through the nodes and get all the variables. 

Question is how do i get variables that are only assigned with a specific "ns". for example, if we need only the variables from ns=1; i=(can be any thing).  

 `node = client.get_objects_node()
print("Starting Node: ", node)

def browse_recursive(node):
    for childId in node.get_children():
        #print ("ChildID :", childId)
        ch = client.get_node(childId)
        if ch.get_node_class() == ua.NodeClass.Object:
            browse_recursive(ch)
            #print ("Recusrising ", ch)
        elif ch.get_node_class() == ua.NodeClass.Variable:
            try:
                print ("Node is variable ", ch)
            except ua.uaerrors._auto.BadWaitingForInitialData:
                pass
    
browse_recursive(node)`

SHAFIT, 2023-02-04:
I was trying to get the complete list of varibales. But it is shoiwng some data which are not really required. 
is it possible to browse only the variables which are assinged to a particular "ns". for example "ns=1; " 

```
node = client.get_objects_node()
print("Starting Node: ", node)

def browse_recursive(node):
    for childId in node.get_children():
        ch = client.get_node(childId)
        if ch.get_node_class() == ua.NodeClass.Object:
            browse_recursive(ch)
            #print ("Recusrising ", ch)
        elif ch.get_node_class() == ua.NodeClass.Variable:
            try:
                print ("Node is variable ", ch)
            except ua.uaerrors._auto.BadWaitingForInitialData:
                pass

browse_recursive(node)
```