FreeOpcUa/python-opcua #866: EventGenerator creates wrong events

korny86, 2019-07-22:
In case a custom event has the same nodeid (but a different namespace) like a standard event, the get_event_generator() function creates an event generator with the standard event.
The problem seems to be in the get_event_obj_from_type_node() in events.py.
[image]

A Minimal Example for testing:
```

import time
from opcua import ua, Server

if __name__ == "__main__":
    server = Server()
    uri = "https://www.randompage.whatever/"
    server.set_application_uri(uri)
    server.set_endpoint("opc.tcp://localhost:1236/MinExample")
    server.set_server_name("MinExample")
    server.set_security_policy([ua.SecurityPolicyType.NoSecurity])

    idx = 1

    base_event_type = server.get_root_node().get_child(["0:Types", "0:EventTypes", "0:BaseEventType"])
    node_id = ua.NodeId(2052, idx)  # 0:2052 is the AuditEventType
    my_new_event_type = base_event_type.add_object_type(node_id, "MyNewEvent")

    myevgen = server.get_event_generator(etype=my_new_event_type)

    server.start()

    while(True):
        myevgen.trigger()
        time.sleep(1)
```

And here is the wrong event shown in uaexpert:
[image]

Kind regards,
Enrico

Waldkaeuzchen, 2019-08-08:
I have exactly the same problem in my project, is there already a solution?

oroulet, 2019-08-14:
probably not. Someone has to look at that code and find out